meshtastic/firmware #7100: [Bug]: 2.7 BaseUI overlaps on the T-Echo

kerosina, 2025-06-21:
### Category

Other

### Hardware

T-Echo

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7

### Description

On the T-Echo with the new BaseUI, some elements overlap each other. For example, the home menu title and the battery percentage, or sometimes the battery percentage and the battery icon.
Additionally, some shortcuts from the old default UI no longer work, like 4x for enabling the backlight. Is there some new key combo for it?
(On a side note, I can't set canned messages when using InkHUD, even though theres a `send` button with what I think are pre-defined messages)

### Relevant log output

```Shell

```

todd-herbert, 2025-06-21:
Just waiting for InkHUD's canned message feature to get merged: https://github.com/meshtastic/firmware/pull/7096

Xaositek, 2025-06-22:
Let me see what I can do with the overlap issue

Xaositek, 2025-06-22:
I have this fixed now in a branch - I will merge this in the coming days.

[image]

closed 2025-06-22 by Xaositek