meshtastic/firmware #7103: [Bug]: FW 2.7.x: Node rebbots at Bluetooth pairing (NRF52840 + RA62)

mrekin, 2025-06-22:
### Category

BLE

### Hardware

DIY

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

v2.7.0.195b7cc

### Description

Hi!

Node with diy promicro fw variant (promicro (nrf52840 + ht-ra62)
Problem exist on v2.7.0.195b7cc and current codebase.
No problems at any 2.6.xx

Node re-boots right after smartphone tries pair via bluetooth 
_- remove node from known bt devices and try to pair again_.
I can successfully connect to node via serial from this smartfone and there are no problems.

Part of log:

```
2025-06-22 15:19:28,230 ~  DEBUG | ??:??:?? 201 [Power] Battery: usbPower=0, isCharging=0, batMv=-1, batPct=0
2025-06-22 15:19:48,238 ~  DEBUG | ??:??:?? 221 [Power] Battery: usbPower=0, isCharging=0, batMv=-1, batPct=0
2025-06-22 15:19:48,278 ~  INFO  | ??:??:?? 221 BLE Connected to OnePlus 7T
2025-06-22 15:19:48,836 ~  INFO  | ??:??:?? 0 No I2C devices found
2025-06-22 15:19:48,847 ~  DEBUG | ??:??:?? 0 acc_info = 0
2025-06-22 15:19:48,853 ~  INFO  | ??:??:?? 0 S:B:63,2.7.0.195b7cc
2025-06-22 15:19:48,857 ~  INFO  | ??:??:?? 0 Build timestamp: 1750515982
2025-06-22 15:19:48,938 ~  DEBUG | ??:??:?? 0 Reset reason: 0x0
2025-06-22 15:19:48,947 ~  DEBUG | ??:??:?? 0 Set random seed 3513881787
2025-06-22 15:19:48,947 ~  INFO  | ??:??:?? 0 Init NodeDB
2025-06-22 15:19:48,947 ~  INFO  | ??:??:?? 0 Load /prefs/nodes.proto
2025-06-22 15:19:48,947 ~  INFO  | ??:??:?? 0 Loaded /prefs/nodes.proto successfully

```

### Relevant log output

```Shell

```

elmanuel127, 2025-06-26:
I can confirm that this also happens on a rak4631 board. I flashed it with 2.7.0 and could not pair the device. It would immediately drop the bluetooth connection after entering the PIN.

The same board could be updated from 2.6.11 to 2.7.0 and it work just fine, but trying to pair a factory reset node on 2.7.0 would result in the device dropping the connection and restarting.

I have noticed that the board emits a high pitched whine immediately after startup on 2.7.0. The same board is silent up until I start pairing process on 2.6.11.

porkcube, 2025-06-27:
Someone dug into the commits [and found the culprit](https://github.com/jeremiah-k/meshtastic-firmware/blob/investigation/pairing-crash-deep-dive/BLE_INVESTIGATION_FINAL_RESULTS.md) - and confirming using a 2.6.13 firmware to pair then immediately updating to 2.7.1 is a valid workaround until it's fixed upstream.