meshtastic/firmware #7110: Should be easier to know all possible config options

chrisportela, 2025-06-23:
Here is where the config is loaded from the config.yaml: https://github.com/meshtastic/firmware/blob/0808f5215ffa3beb8240349170eca65dfadb5b24/src/platform/portduino/PortduinoGlue.cpp#L444

It should be documented or at least defined somewhere what options are allowed in the Yaml config.

It didn't solve my current problems, but in trying to debug my meshtasticd configuration this was harder than it needs to be.

kawfey, 2025-06-27:
i would be interested where all possible yaml config parameters differ from https://meshtastic.org/docs/configuration/

chrisportela, 2025-06-27:
They're very different. The config.yaml is very limited and strict. It helps configure meshtasticd with things like the pins for I2C or which radio you are using. 

Most of the options listed on those pages either aren't configurable via the config.yaml or they differences in how you describe them.

For instance, the display configuration doesn't let you set the type of OLED panel like you can via the web interface or mobile app(s) (which helps set the values you linked). It just isn't an option. At most you can specify which I2C linux device to use. Instead it does some auto detection by reading the I2C memory. Another example: you also can't preset the region you're using (or so it seems to me).