meshtastic/firmware #7135: [Bug]: Heap climbs to 99 % → BLE unusable on T-Beam Supreme L76K

Boa-Thomas, 2025-06-25:
### Category

Other

### Hardware

T-Beam S3

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.0195b7cc

### Description

Board
LilyGO T-Beam Supreme L76K (ESP32-S3, SX1262, 16 MB flash, 8 MB PSRAM)

Firmware
2.7.0.195b7cc (tech-preview, 25 Jun 2025)
But I seems to rebember experiencing this issue on older firmware also!

Steps to reproduce

1. Flash the build above and factory-reset (erase config).
3. Boot and run get sys heap → (~60 % used).
5. Open the TFT menu System → System once; re-run the command.
7. Wait up to 2 days. (My device uptime is 2d8h at the time of the picture) Heap free memory drops.
9. At 99 % used try to pair the Meshtastic app → fails;

Expected
Heap remains <80 % used and BLE stays available for the whole uptime.

Actual
Heap leaks until 99 % used; BLE stack refuses to start, node becomes undiscoverable. Power-cycle is the only recovery.

[image]

### Relevant log output

```Shell

```

Boa-Thomas, 2025-06-25:
Here are some logs of me trying to connect using the Android app

`[0m[32mINFO  [0m| 17:25:51 203343 [UserButton] [32mInput event 28! kb 0
[0m[32mINFO  [0m| 17:25:53 203344 [UserButton] [32mInput event 28! kb 0
[0m[32mINFO  [0m| 17:25:54 203345 [UserButton] [32mInput event 28! kb 0
[0m[32mINFO  [0m| 17:25:55 203346 [UserButton] [32mInput event 28! kb 0
[0m[32mINFO  [0m| 17:25:55 203346 [UserButton] [32mInput event 28! kb 0
[0m[34mDEBUG [0m| 17:25:56 203347 [Power] [34mBattery: usbPower=1, isCharging=0, batMv=4114, batPct=100
[0m[32mINFO  [0m| 17:25:59 203350 [32mBLE authentication complete
[0m[34mDEBUG [0m| 17:25:59 203350 [34mBluetoothStatus CONNECTED
[0m[32mINFO  [0m| 17:26:03 203354 [DeviceTelemetry] [32mSend: air_util_tx=1.310167, channel_utilization=1.863333, battery_level=100, voltage=4.114000, uptime=203354
[0m[34mDEBUG [0m| 17:26:03 203354 [DeviceTelemetry] [34mPartially randomized packet id 1281290508
[0m[34mDEBUG [0m| 17:26:03 203354 [DeviceTelemetry] [34mupdateTelemetry LOCAL
[0m[34mDEBUG [0m| 17:26:03 203354 [DeviceTelemetry] [34mNode status update: 1 online, 3 total
[0m[32mINFO  [0m| 17:26:03 203354 [DeviceTelemetry] [32mSend packet to phone
[0m[32mINFO  [0m| 17:26:03 203354 [DeviceTelemetry] [32mSending local stats: uptime=203354, channel_utilization=1.863333, air_util_tx=1.310167, num_online_nodes=1, num_total_nodes=3
[0m[32mINFO  [0m| 17:26:03 203354 [DeviceTelemetry] [32mnum_packets_tx=13758, num_packets_rx=13686, num_packets_rx_bad=0
[0m[34mDEBUG [0m| 17:26:03 203354 [DeviceTelemetry] [34mPartially randomized packet id 3014155533
[0m[32mINFO  [0m| 17:26:03 203354 [32mTell client we have new packets 24674
[0m[34mDEBUG [0m| 17:26:03 203354 [SerialConsole] [34mFromRadio=STATE_SEND_PACKETS
[0m[34mDEBUG [0m| 17:26:03 203354 [SerialConsole] [34mphone downloaded packet (id=0x4c5ef10c fr=0x43575750 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1750883163 priority=10)
[0m��`

[image]

HarukiToreda, 2025-06-26:
I've had a similar issue with the original UI, it happens when you update the node many times without doing a full wipe, after a full wipe the heap becomes stable.

Boa-Thomas, 2025-06-29:
> BaseUI just lets you visually see what was out of view before, not a baseUI created issue.

Agreed, will do a full wipe and see if it solves it.

Boa-Thomas, 2025-06-29:
Even after a full wipe and reinstall of the firmware is still climbs...
21,5h of uptime
[image]

closed 2025-09-29 by github-actions[bot]

Boa-Thomas, 2025-09-29:
any fix in view? still happens as version 2.7.4 (c1f4f79)