meshtastic/firmware #7141: [Bug]: Raspberry Pi 4 + Seeed Gateway WM1302 (only using its SX1262)

ToshikSoni, 2025-06-26:
### Category

Hardware Compatibility

### Hardware

Linux Native

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.11.60ec05e Beta

### Description

Hello, I am trying to use Seeed Gateway's SX1262 chip with Raspberry Pi 4. My pin config is cs=6, irq=23, rst=5, busy=18. Which I already tested with RadioLib's Raspberry Pi example and it is working fine. I cannot understand why it is not initializing with Meshtastic. And yes, I know WM1302 is having SX1302 but there is a separate SX1262 also there, which is used for LBT, but I am not using SX1302 so I can independently use the SX1262 without any issue. I have tested this using the SPI pins present; here is an image with the pin diagram. I have tested its SX1262 using RadioLib and sent and received packets from Heltec LoRa32 v3.2. 

[image]

Here is the example code I used:
```
#include <RadioLib.h>
#include "hal/RPi/PiHal.h"
#include <string>
PiHal *hal = new PiHal(1);
SX1262 radio = new Module(hal, 6, 23, 5, 18);
int main(int argc, char **argv)
{
    printf("[SX1262] Initializing ... ");
    int state = radio.begin(868.0, 125.0, 9, 7);
    if (state != RADIOLIB_ERR_NONE)
    {
        printf("failed, code %d\n", state);
        return (1);
    }
    printf("success!\n");
    int count = 0;
    for (;;)
    {
        printf(("[SX1262] Transmitting packet ... "));
        printf(("Count: %d ... "), count);
        std::string str = "Hello World! From Seeed SX1262#" + std::to_string(count++);
        uint8_t *data = new uint8_t[str.length() + 1];
        for (int i = 0; i < str.length() + 1; i++)
        {
            data[i] = str[i];
        }
        int state = radio.transmit(data, str.length());
        if (state == RADIOLIB_ERR_NONE)
        {
            printf(("success!\n"));
        }
        else if (state == RADIOLIB_ERR_TX_TIMEOUT)
        {
            // timeout occured while transmitting packet
            printf(("timeout!\n"));
        }
        else
        {
            printf(("failed, code "));
            printf("%d\n", state);
        }
    }
    return (0);
}
```
If it is working with RadioLib, then theoretically it should work with Meshtastic too.
Also, I am new to Meshtastic and might have missed something. I would really appreciate it if someone could guide me.

### Relevant log output

```Shell
DEBUG | 04:56:53 0 Activate sx1262 radio on SPI port /dev/spidev0.0
DEBUG | 04:56:53 0 SX126xInterface(cs=6, irq=23, rst=5, busy=18)
DEBUG | 04:56:53 0 SX126X_DIO3_TCXO_VOLTAGE not defined, not using DIO3 as TCXO reference voltage
INFO  | 04:56:53 0 Start meshradio init
INFO  | 04:56:53 0 Radio freq=868.375, config.lora.frequency_offset=0.000
INFO  | 04:56:53 0 Set radio: region=UNSET, name=LongFast, config=0, ch=3, power=14
INFO  | 04:56:53 0 myRegion->freqStart -> myRegion->freqEnd: 867.500000 -> 868.500000 (1.000000 MHz)
INFO  | 04:56:53 0 numChannels: 4 x 250.000kHz
INFO  | 04:56:53 0 channel_num: 4
INFO  | 04:56:53 0 frequency: 868.375000
INFO  | 04:56:53 0 Slot time: 28 msec
INFO  | 04:56:53 0 Final Tx power: 14 dBm
INFO  | 04:57:03 10 SX126x init result -2
WARN  | 04:57:03 10 No sx1262 radio
```

caveman99, 2025-06-26:
The SX1261 (!) in these chips is not hooked up with the TX switch. It's RX only. So even if you could get it to be recognized you won't get any range out of it.

ToshikSoni, 2025-06-26:
Hey, thanks for your response. I have tested the Seeed Gateway's SX1262 and it is working fine for me. I used Heltec LoRa v3.2 to test it and I am able to transmit and receive the data. Here are the screenshots. The RSSI and SNR are not working as of now but the rest is working. I know it is an odd configuration, but it should be possible, as it can work on RadioLib.

I am trying to use the PiHal.h from the example code of RadioLib. here is the link: https://github.com/jgromes/RadioLib/blob/master/examples/NonArduino/Raspberry/main.cpp

Probably it should work, please let me know if there is anything else I can try.

[image]

[image]

ToshikSoni, 2025-06-28:
Ok, my apologies. I just tested that in the lab. The range is just up to 3 or 4 meters. Previously the devices were just beside each other. Thank you for your time.

closed 2025-06-28 by ToshikSoni

GUVWAF, 2025-06-28:
No problem, it's a shame the Tx side is not connected.