meshtastic/firmware #7149: [Feature Request]: UDP support for RAK13800 ethernet module

kabili207, 2025-06-27:
### Platform

NRF52

### Description

I would really love to be able to use the Mesh over UDP feature with this ethernet module. I did see possible support was mentioned in a comment on [the original pull request](https://github.com/meshtastic/firmware/pull/5779#issuecomment-2577207713), but it doesn't look like it has been mentioned anywhere else.

I wasn't able to find a recent, maintained version of AsyncUDP for this module, so I get the feeling this will require a very different approach than the ESP32 and Linux native ports

vidplace7, 2025-06-30:
The UDP implementation on ESP32 relies on the [`AsyncUDP`](https://github.com/espressif/arduino-esp32/tree/master/libraries/AsyncUDP) library. Unfortunately there's no such library for the nRF52 series.
TCP/IP networking is really an afterthought on the [Arduino-nRF52](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries) platform. ☹

thebentern, 2025-07-20:
Closed with #7385

closed 2025-07-20 by thebentern