meshtastic/firmware #7157: [Feature Request]: Support NMEA output on USB-CDC instead of Debug (and more)?

jstockdale, 2025-06-27:
### Platform

Cross-Platform

### Description

Hi! I have been struggling with how to frame this, but basically the USB-CDC connection and the serial output are used in mutually exclusive ways to inform users of the state of the device. I think it would make a bunch of sense to unify the output streams across the different ports and allow the user to select different combinations.

Right now we have–
USB-CDC: Full device debug log (not available anywhere else)
Serial: Dumb terminal, Text messages out, NMEA messages, CalTopo messages, Protocol Buffer

I've wanted to get the Full device debug log to come out of the serial port many times. I created a serial log display with ansi color code support out of a T-Display-S3-Long unit. But in order to get the debug log that's coming out of the USB-CDC port out of the serial port, I had to create a custom branch of the code to override how the serial ports are set up and what data is output where. This is well beyond most users' capability.

I've also often wanted to configure a device to output NMEA sentences over USB-CDC from the GPS, so it can be used as a usb serial device for a laptop or desktop computer. This is not supported. You can output NMEA over serial but then you'll also need a usb to serial adapter or similar. I've set up many of my devices to use Launcher and dual boot ESP Marauder as it does have a raw NMEA gps output mode that outputs over USB-CDC ... and it works great.

I'm requesting that abstract the output port and the information streams in a manner that lets the user select any output for the USB-CDC and any output for the Serial port(s) on the device. So you can have any data coming out of any port or the same data coming from all three.

I'm happy to help with the implementation if helpful. But I wanted to have a discussion before unilaterally implementing this in a PR.

Thanks all!

-John

thebentern, 2025-06-27:
Have you tried this config?
https://meshtastic.org/docs/configuration/module/serial/#override-console-serial-port

The primary use case was to play NMEA sentences over the USB, particularly for devices like the T-Echo which have onboard GNSS but no UART port.

jstockdale, 2025-07-04:
Oh hey! That's super useful. Let me test it out. Thank you :-)