meshtastic/firmware #7160: [Feature Request]: 2.7.1 option to disable beeping with unhealthy IAQ

ponzano, 2025-06-28:
### Platform

NRF52

### Description

We’re having very hot weather in Italy and air quality in my city (Turin) isn’t the best. My node running 2.7.1 has been beeping every minute since yesterday… I know air quality is Very Unhealthy but there isn’t much I can do about it 😂😭
There should be an option to disable this beeping related to IAQ.

alexsavuto, 2025-06-28:
Have you tried, on the system screen (if you have a screen) holding down the button?

ponzano, 2025-06-29:
> Have you tried, on the system screen (if you have a screen) holding down the button?

Yes I had forgotten about those options 😅
For some reason it would beep even while on other pages but after a reboot it seems fixed 🤔
Probably just a bug, if it only beeps while on that page, then I think adding an option to disable it isn't a big priority

fifieldt, 2025-07-28:
OK, closing for now - please reopen if it starts happening again. A lot of work went into the screen in the past month, so hoping it is fixed :)

closed 2025-07-28 by fifieldt