meshtastic/firmware #7173: [Feature Request]: Legal Remote Admin for Radio Amateurs.

MikeMesh, 2025-06-30:
### Platform

NRF52

### Description

In the android app, as soon as you identify yourself as a radio amateur, encryption is removed. Once this happens we can no longer access remote admin features. 

As a radio amateur, we need a solution to this issue that keeps within the bounds of the law. Maybe a specialized private channel set up? 

Thanks for considering this issue. 

Cheers

MikeMesh, 2025-06-30:
It is clearly stated within our telecommunications act that a licensed radio amateur may not engage in encryption where the key is secret. At any level of power. 

**Section 47 A person who operates radio apparatus in the amateur radio service may only . . .  (b) use a code or cipher that is not secret”**

Encryption is prohibited in both the U.S. and Canada.

Of course we can use Morse code, RTTY AMTOR etc.... this form of digital encryption and its key are known to all hence not secret. 

So, It would be nice to find away to operate within the bounds of the law. Not stating the fact you are a licensed operator governed by this act would be a contravention thereof. 

lets find a legal path forward shall we?

Regards

pdxlocations, 2025-06-30:
> the moment you got your license you are not allowed to use meshtastic encrypted anymore

This is untrue.  licensed mode is voluntary.  Hams are allowed to use unlicensed systems if they follow the rules that apply to non-hams.

MikeMesh, 2025-06-30:
I think what everyone is misunderstanding is the the band we operation in is the 33 cm band. This is not an ISM band to a licensed operator. It is a prescribed band for licensed people. Hence the rules apply whether you are running 10mW or 10W. You will find no cut out for licensed people to run encryption. 

The ARRL is already beginning to structure the band. 
https://www.arrl.org/files/media/News/[33_cm_Band_Plan-Draft.pdf](https://www.arrl.org/files/media/News/33_cm_Band_Plan-Draft.pdf)

MikeMesh, 2025-06-30:
> > Please site the regulations in Canada and the US that clearly depict this.
> 
> Apologies as I am no expert by any stretch of the word but I'm guessing there are no "regulations" for hams operating as non-hams. Baby monitors, doorbells, solar/ev chargers etc. may all use this band and may be encrypted. Would I be in violation as an amateur operator if I used these devices?

There is no "cut out" that says you can. It also depends on the band. The 33 cm band is within our area of licensed operations. Likely baby monitors and cordless phones have there own unregulated band sections not assigned to licensed operators. I dont see cut out for any of those devices within the amateur bands.  

There is only one cut out in the entire area of amateur operations that permits encryption. That is satellite control. Radio amateurs for years have run their own sats. They negotiated a cut out to control those birds. But that cut out is strictly relegated to satellite operations. 

Bottom line, when I was granted my license, I signed on the dotted line I would adhere to the regulations. Within those regulations there is no specific statement granting a licensed operator the right to run encrypted within prescribed band plans set out for us.

Herein lays the problem. An oversite in today digital world? Probably. I think transmitted and encrypted password only for the direct management of a system would not ruffle any feathers.

pdxlocations, 2025-06-30:
@MikeMesh would you allow yourself to use any products that leverage LoRa or LoRaWAN?

This is an interesting topic and the bottom bottom line is we want everyone to get the most enjoyment out of Meshtastic as possible. 

One user on Discord did find an interesting regulation that could be interpreted to mean using ham mode is prohibited!
https://www.ecfr.gov/current/title-47/chapter-I/subchapter-D/part-97/subpart-B/section-97.113#p-97.113(a)(5)

MikeMesh, 2025-07-12:
> [@MikeMesh](https://github.com/MikeMesh) would you allow yourself to use any products that leverage LoRa or LoRaWAN?
>
> This is an interesting topic and the bottom bottom line is we want everyone to [...]

Thanks for the comment. None of this applies in my country of Canada. Also, they are misinterpreting section 5.

pelgraine, 2025-08-19:
Just to tack onto this I also requested this be done earlier as remote administration is legal for licenced hams in Australia and NZ, and the request was closed. I still don't understand why.

More details here: https://github.com/meshtastic/firmware/issues/7401

jbaenaxd, 2025-10-18:
What about setting a OTP? It will be broadcast with no encryption and the moment that the nodes receives it, it will change to non HAM mode, so it can let other nodes to send their administration instructions.

Also, the problem is broadcasting, not receiving, your HAM node should be able to receive encrypted administration commands, only that it should only reply in non HAM mode.