meshtastic/firmware #7184: [Feature Request]: Chatter2 button change and clock update

eureekasigns, 2025-07-01:
### Platform

ESP32

### Description

Not sure best way to request this so if this isn't it I would take no offense. 
2.7 changes navigation.

To use Chatter2 keys for navigating, one way that seems to work well is to remap the RIGHT button to be DOWN. Otherwise, it's not possible to fully navigate the menus. 

Line 90 here 
https://github.com/meshtastic/firmware/blob/master/src/input/SerialKeyboard.cpp

Backspace key functions to navigate main menus.

Not sure if it's worthwhile or possible to add a way to use shift key to change the UP/DOWN keys to change to LEFT/RIGHT functions. If so, perhaps that is another way but may require more work. 

If next request should be separate, I do not know. 

Clock display appears fixed when changing the following as noted by another user here
https://github.com/meshtastic/firmware/discussions/6896#discussioncomment-13611071

ClockRenderrer.cpp (https://github.com/meshtastic/firmware/blob/master/src/graphics/draw/ClockRenderer.cpp) in line 237, so I tried to modify the code from the original

237 scale = 1.5;

to this

237 scale = 1.1;

Xaositek, 2025-07-03:
@eureekasigns - looking at this a bit.

Working branch has been created to put any fixes against.

Branch: https://github.com/meshtastic/firmware/tree/chatter_2_fixes
Draft PR Here: https://github.com/meshtastic/firmware/pull/7214/files

eureekasigns, 2025-07-03:
Thank you @Xaositek

eureekasigns, 2025-07-04:
On the serial keyboard change, I'm not sure we need "tab" function any longer, fyi 

Destinations can be selected with down button, being active if we change that. Then, if you click cancel and start typing it will be to the last selected destination, whether node or channel.

Adding shift function to get all 4 directions working may be more work. Changing right to say "down" on one line instead is a very small change that allows functionality. Hoping that's possible, or if there's some way that's easy to do all. Thought it would be simpler to get basic function first. Thanks again.

eureekasigns, 2025-07-23:
> we got some CHATTER 2.0 nodes coming our way soon so we can test how things work and possibly fix it.

That's exciting! Looking forward to seeing how you think would be best for the navigation. 

For functionality, I am ok with the one word change that I was able to find, though I am not much of a programmer. 

It may be better to have shift change the arrow direction from left right to up down. Then maybe also remove whatever was used for the "tab" function that worked in the older firmware versions. 

In either case, thank you for taking a look!

eureekasigns, 2025-08-12:
Hello @HarukiToreda - just wondering if you'd had any time to look into this and also interested in thoughts on either the 1 line change or rework for all directions/removing "tab" function.

HarukiToreda, 2025-08-13:
unfortunately I have been unsuccessful to even get the firmware loaded on it so no progress at all. Meshtastic won't even initiate.

[image]

> Hello [@HarukiToreda](https://github.com/HarukiToreda) - just wondering if you'd had any time to look into this and also interested in thoughts on either the 1 line change or rework for all directions/removing "tab" function.

eureekasigns, 2025-08-13:
Ah dang. It is a little tricky to short the correct pin on the chip to ground as it powers on to get it into flashing mode. This gets it to a state that it can be flashed using CLI (or VSC). 

A short thin wire can work fine, though it's a little easier if soldering to either a momentary button or to one on the board. 

To start, I just used a loose wire - holding it on the cou pin+a ground point like where a screw goes. Power on and it should have a blank screen. Then run the flash command. Then I soldered to just the CPU pin so I could tap to ground. Then wired a button later.

I believe the thread had info on that - bottom right pin to left pad of bottom right button is one way
https://github.com/meshtastic/firmware/issues/2896#issuecomment-2384343529

eureekasigns, 2025-08-29:
If no luck yet, can we maybe look at getting the one line approved to be changed to at least get baseUI to function? 
It'd be fun to have full navigation but having it functional would be a great start and super easy PR.

eureekasigns, 2025-09-27:
@HarukiToreda can we get "down" arrow key set at some point please? 

Sorry you couldn't get yours to flash. Hopefully you're up for trying again at some point. They are fun devices with Meshtastic.

Also saw latest talk on multi message storage and building that to try!  Looks really great. Thank you for those efforts BTW!

HarukiToreda, 2025-11-12:
PR Opened with all fixes, would appreciate some testers
https://github.com/meshtastic/firmware/pull/8615

closed 2025-12-29 by HarukiToreda