meshtastic/firmware #7232: [Bug]: Uplinked messages from LoRa nodes not reaching MQTT

softsmyth, 2025-07-04:
### Category

Other

### Hardware

Not Applicable

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.11.60ec05e

### Description

My setup consists of two meshes  linked by MQTT:   [B] <-> [A] -- MQTT -- [1] <-> [2] ..... 

- messages originating from [A] are seen on [1] or [2]. messages originating from [1] are seen on [A] or [B]. In this case, while monitoring via MQTT Explorer, I can see the posts from the originating MQTT enabled node and then a few seconds later a response to those messages posted by the opposite MQTT enabled node. 
- messages from [B] are not seen by [1] or [2] AND they not posted in MQTT. messages from [2] are not seen by [A] or [B] AND they not posted in MQTT. In this case, while monitoring via MQTT Explorer, I don't see ANY posts from the originating MQTT enabled node that should be passing the message to MQTT.

I'm using a named private channel on index 0 with generated PSK shared among all the nodes and no secondary channel. I have a private MQTT broker running. I have 'OK to MQTT' enabled in ALL nodes as well as both uplink and downlink enabled in the private channel config for ALL nodes. Channel name and PSK are identical in all nodes. I'm running 2.6.11 in all nodes which are a variety of both Heltek and Liliygo. Using a private channel and MQTT server eliminates the zero hop count issue...

From my experiments, I know the MQTT server is working (it has other active uses as well) and I'm seeing posts from the MQTT module enabled nodes on both meshes. What I'm not seeing is the local nodes (both meshes) unable to target a message to any node in the opposite mesh, which I can see via MQTT explorer, are not being forwarded via the MQTT enable node of that mesh. There is something preventing the forwarding of out-going messages via the MQTT module enabled node. 

### Relevant log output

```Shell

```

softsmyth, 2025-07-10:
I have attached four txt files containing --info for nodes A, B, 1, and 2

[node-1.txt](https://github.com/user-attachments/files/21166363/node-1.txt)
[node-2.txt](https://github.com/user-attachments/files/21166362/node-2.txt)
[node-A.txt](https://github.com/user-attachments/files/21166360/node-A.txt)
[node-B.txt](https://github.com/user-attachments/files/21166361/node-B.txt)

GUVWAF, 2025-07-12:
Hmm, I can't figure out why this wouldn't work, your settings look correct. 
Would you be able to check the serial logs from node A when sending a packet from either B to 1, or 1 to B? You can use the "Open Serial Monitor" button in the [flasher](https://flasher.meshtastic.org/) or use the Python CLI command `--noproto`.

closed 2025-09-04 by github-actions[bot]