meshtastic/firmware #7242: [Feature Request]: Launch Meshtastic with Dragino SX1276 LoRa HAT v1.4 (Linux Native)

kdHub, 2025-07-06:
### Platform

RP2040

### Description

Setup

Device: Dragino LoRa HAT v1.4 (SX1276), Pi running rasbian 12 with meshtasticd installed successful 

OS: Linux (Native, not Docker or Pi image)

Interface: SPI (/dev/spidev0.0 is present)

Tool: meshtasticd

Problem
I'm trying to use meshtasticd with my Dragino SX1276 LoRa HAT v1.4 but can't get it to launch correctly. The SPI device (/dev/spidev0.0) is visible, so I believe the hardware is recognized by the system.

Question
Is the Dragino SX1276 HAT v1.4 a supported board for Meshtastic using the Linux Native driver (meshtasticd)? If yes, are there any specific steps, overlays, or configurations required to get it working?

What I’ve Tried:

Confirmed SPI is enabled and /dev/spidev0.0 exists

Reviewed available documentation and GitHub issues

Any tips or guidance would be greatly appreciated.

config.yaml example:
Lora:
  Module: sx1276
  CS: 7
  IRQ: 22
  Reset: 11

Link to board diagram Im using:
https://github.com/dragino/Lora/blob/master/Lora_GPS%20HAT/v1.4/Lora%20GPS%20%20HAT%20for%20RPi%20v1.4.pdf

lyusupov, 2025-07-06:
[image]

The HAT is for Raspberry Pi which essentially uses **Broadcom** BCM283X hardware platform. 
It has nothing to do with RP2040.

Dragino Tech Support had reported few years ago that **they no longer manufacture this device** since late years 201x.
I believe, It has no sense to spend time on a board that is not available for sale anymore...

Unlike the deprecated Dragino product mentioned in the report,
 this product from Waveshare is still available on the market:
https://www.waveshare.com/sx1262-lorawan-hat.htm

[image]

[image]

kdHub, 2025-07-06:
Thanks for the clarification. I understand this is a legacy product—I had to select something from the dropdown to submit the issue, so apologies if that was misleading.

I'm primarily trying to find out whether anyone has successfully gotten this working, or if it's simply a matter of getting the pin values correct. I'm still learning the Meshtastic system and don’t typically use Discord, so I appreciate any guidance.

I'm mostly interested in understanding what it would take to get this module supported, even if it's no longer in production—mainly as a learning exercise. Thanks!