meshtastic/firmware #7246: [Bug]: T114 Color display selection disables BT after reboot

LostSoul3471, 2025-07-06:
### Category

Hardware Compatibility

### Hardware

Heltec Mesh Node T114

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.x

### Description

1st off, leaps and bounds better UI than v2.6 but one minor issue is that if you select color display option on Auto-OLED it will boot to blank screen and disable BLE in the config.  You can get back into it by connecting via serial and then re-enabling BLE support, but it's been a little broken since the 2.7 preview release.  Not a crippling error as you can get back in so long as you can connect via serial but could be an issue for those that can't.

### Relevant log output

```Shell

```

bplein, 2025-07-06:
I ran into this today as well, Heltec T114 v2.  So count that as one other person reproducing the issue, on `2.7.1.f35ca81`

Side question: Does the color display work at all with Meshtastic?

Xaositek, 2025-07-06:
The T114 will attempt to boot MUI (which it doesn't have) if you select "COLOR" display in the app. Set the display to "Default 128x64 screen layout" in the app and then manage the color on the device itself through the menu system.

PR #7248 (2025-07-06): Automatically bail user out of displaymode_color when not HAS_TFT
Fixes #7246
closed 2025-07-06 by jp-bennett