meshtastic/firmware #7256: [Bug]: ANZ433 and MY433 regions allow frequency outside the permitted range

NomDeTom, 2025-07-07:
### Category

Other

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

???

### Description

In #7036 the ANZ433 range was added, which changed the tx power limits. I note that it states the lower frequency bound (433.05) correctly, but it allows 8 frequency slots in the range, starting at 433.00 (edge of slot, below the permitted range) which takes it to 434.875 (centre frequency) which is 435.00 (edge of slot) and over 434.79, which is the upper limit.

If the correct range is used, there should only be 6 slots.

### Relevant log output

```Shell

```

NomDeTom, 2025-07-10:
@fifieldt I tried tracking down how it is used, but it is packaged up and sent away pretty quickly. I'm away from my hardware right now, and I'm not sure if the frequency range that the android app uses is even real.

I'll try and look at it this weekend.

fifieldt, 2025-07-16:
Looking at the code ... 

    RDEF(ANZ_433, 433.05f, 434.79f, 100, 0, 14, true, false, false),
    
    
    bw = (myRegion->wideLora) ? 812.5 : 250;

    uint32_t numChannels = floor((myRegion->freqEnd - myRegion->freqStart) / (myRegion->spacing + (bw / 1000)));

     uint32_t numChannels = floor((434.79f - 433.05f) / (0 + (250 / 1000)))

so numChannels *should* be 6 ... but due to bw and 1000 both being ints, it's not?

need a log to check

fifieldt, 2025-07-17:
```INFO  | 00:59:47 0 Start meshradio init
INFO  | 00:59:47 0 Radio freq=434.425, config.lora.frequency_offset=0.000
INFO  | 00:59:47 0 Set radio: region=ANZ_433, name=LongFast, config=0, ch=5, power=14
INFO  | 00:59:47 0 myRegion->freqStart -> myRegion->freqEnd: 433.049988 -> 434.790009 (1.740021 MHz)
INFO  | 00:59:47 0 numChannels: 6 x 250.000kHz
INFO  | 00:59:47 0 channel_num: 6
INFO  | 00:59:47 0 frequency: 434.424988
```

fifieldt, 2025-07-17:
so

6 * 250kHz < 1.74 MHz

so it should be OK

weird that it sets 

 433.049988

rather than

 433.05

fifieldt, 2025-07-21:
Fixed by https://github.com/meshtastic/Meshtastic-Android/pull/2466

closed 2025-07-21 by fifieldt