meshtastic/firmware #7274: [Bug]: T114 does not turn on external OLED

Deuteranomalous1, 2025-07-09:
### Category

Hardware Compatibility

### Hardware

Heltec Mesh Node T114

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.1

### Description

Reporting as per HarukiToreda on discord. https://discord.com/channels/867578229534359593/925629468289171457/1392199424993661019 

Connecting 2.42" OLED to T114 results in no image on the screen. https://vi.aliexpress.com/item/1005002766729064.html
Same OLED module with same CardKB to a RAK 19007 works as expected. HarukiToreda thinks this OLED should work on the T114 as well. 

OLED SDA/SCL are connected to pins 0.16 and 0.13 on the T114. The T114 is recognizing the CardKB in the serial output and cardkb can send input. Manually set OLED to OLED_SSD1306 which is the same setting the RAK used. 

Added pics of test setup. Of course whole screen will not display in picture with RAK connected but you can see it is outputting. Totally dark on T114.
Added logs from RAK and T114 respectively. OLED and CardKB are the same units for both logs. 

[image]
[image]

[RAKTXT_COM29-20250708-160826.log](https://github.com/user-attachments/files/21132474/RAKTXT_COM29-20250708-160826.log)
[T114TXT_COM32-20250708-164052.log](https://github.com/user-attachments/files/21132475/T114TXT_COM32-20250708-164052.log)

### Relevant log output

```Shell

```

peterhanneman, 2025-07-24:
I just tried v2.7.3 on a T114 v2 and the screen turned on but had the same issue with 2.7.1.  That said there is another problem with the T114 v2 on v2.7.3 that I'm about to open an issue for where it generates a random bluetooth pin no matter what and can not be set using the CLI.

jnoakes2, 2025-07-24:
> I just tried v2.7.3 on a T114 v2 and the screen turned on but had the same issue with 2.7.1. That said there is another problem with the T114 v2 on v2.7.3 that I'm about to open an issue for where it generates a random bluetooth pin no matter what and can not be set using the CLI.

And that was with an external OLED? I will have to grab another OLED and test!