meshtastic/firmware #7280: [Feature Request]: Define behavior criteria for automatically waking screen on new messages

thebentern, 2025-07-09:
### Platform

Cross-Platform

### Description

We need to define behavior for waking the screen on new messages by default.
As a part of this change, we should make the screen timeout much more aggressively short, perhaps 90-120 seconds. This notification can be triggered by both text messages and waypoints.

There should be several sensible guards are this functionality
- If any ext. notifications are turned on, don't wake screen as the other notifications serve as a call to button press which will wake the screen anyway
- If role is not client / client_mute probably shouldn't wake screen
- If the battery level is very low (perhaps <10%), consider not waking the screen

Xaositek, 2025-07-09:
I am considering managing this as a bool value to take advantage of some code that was written for a protobuf.

The bool should start as **"screen_wake" = true** and then based upon a series of calculations, this becomes **false**.

There have already been plumbing created in this PR that would make some of this a fairly easy swap.

Xaositek, 2025-07-09:
In the branch **screen-wakeup** is now plumbed in a check for if moduleConfig.external_notification.enabled; this will set wake_on_received_message to false.

The function `bool shouldWakeOnReceivedMessage()` can be expanded now with additional conditions on when not to screen wake.

closed 2025-07-22 by Xaositek

ale-rinaldi, 2025-10-23:
@Xaositek do you think it would be possible to add an explicit user setting, like "Never wake up on message" in screen settings? I have a node that has a soldered screen. I don't feel like removing it because sometimes it's useful to push the button and see what's going on, but for the rest of the time it just sits abandoned and externally controlled, so every screen wake is just a power waste.
EDIT: I see I could just enable external notifications without any additional setting, and I would be basically obtaining this, but it looks less clean to me that an explicit "I don't want my screen to turn on unless I push that damn button" :)