meshtastic/firmware #7288: [Feature Request]: Dynamic role switcher

techycpa, 2025-07-10:
### Platform

Cross-Platform

### Description

We have seen many questions about what device role to use with most suggestions to select CLIENT. If each node could automatically change its role type based on changing conditions could improve the stability of the network. I’d propose a node type of AUTO which is set as default and dynamically adjusts.  

Example, a node normally located on the edge of the mesh travels into a high density coverage area. Setting to CLIENT MUTE may make sense as it won’t need to retransmit messages. Another example might be a node set as ROUTER goes offline. Another well placed node that is CLIENT or ROUTER LATE might get promoted to ROUTER to stabilize the mesh.

grleblanc, 2025-07-10:
+1, I really like this idea.

fifieldt, 2025-07-10:
It's a nice idea, but probably pretty hard to implement. Especially to the point it'd be a reasonable default, which I assume is the aim.

Consider eg that a city-wide mesh is just one kind of use case. Also consider that someone going hiking up the nearby mountain doesn't necessarily want their personal device to drain battery faster than before. Consider also devices that are a little "deaf" compared to their "shout".

We already do some scaling automatically based on number of nodes seen, and the air utilization. Retransmissions are already attempted or not attempted based on what other transmissions are seen.

I think to make this idea reality, the next step would be to test out some approaches in the simulator:  https://github.com/meshtastic/Meshtasticator

If there are significant benefits in there, then we might have a winner.