meshtastic/firmware #7291: [Feature Request]: Ability to delete or clear rangetest.csv from device

philoyche, 2025-07-10:
### Platform

ESP32

### Description

When using the range test feature on a esp32 Meshtastic node (e.g., Heltec V3), the results can be saved to a rangeTest.csv file on the internal filesystem. However, there is currently no way to delete or clear this file from the Meshtastic Web UI, the CLI, or the mobile app.

This would be especially helpful for users performing multiple tests or switching deployment environments, as old data may be misleading. It would also reduce the need for low-level manual file system operations.

Thank you for all the amazing work on Meshtastic!

ford-jones, 2025-08-09:
Hey I haven't contributed here before but this looks like something I could work out!
Would submission of this PR require changes to the GUI's to be completed at the same time ?

ford-jones, 2025-08-10:
I've opened a PR to realise this feature-request here: #7595 
If somebody's able to give it a label / review that'd be great 🌐