meshtastic/firmware #7299: [Bug]: Client mute not acknowledging other client mute nodes

willp618, 2025-07-11:
### Category

Hardware Compatibility

### Hardware

T-Beam, Heltec V3, Seeed Card Tracker T1000-E

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.5.15(tbeam) , 2.6.11(t1000e), 2.7.1(heltec v3)

### Description

Currently away on holiday and have been fighting the line through cloud all week been messing with modes and found that if a mesh is made up of just client mute nodes there is no longer an ack reply to messages resulting in the line through the cloud however the message still gets through as soon as the nodes are changed to client it works fine getting ticks in the cloud I'm not sure if it's firmware issues or if it's the android app as all devices are using the update app but with various firmwares we chose client mute as we are all within 400m and have very good signal so don't see why client would be nessory 

### Relevant log output

```Shell

```

b8b8, 2025-07-11:
That's as designed because the ACK you get is actually hearing your own message being rebroadcast. If everyone is on mute, then no one will rebroadcast.

ACKs still work for DMs though on mute since those are targeted to a specific device.

willp618, 2025-07-11:
Im sorry but I think there should be a acknowledgement even if client mute it's caused alot of repeated sending of messages due to us thinking that messages were not being received as we were constantly getting the line through cloud or at least a different icon in the apps for a client mute acknowledgement

willp618, 2025-07-11:
I understand now however there might exist a scenario where group is in range of a existing mesh so they are client mute but then become out of range of a existing mesh in which client mute become an issue they also might not be able to to change modes if they are using homemade stand alone nodes imo client mute needs to either ack a message or the apps need updating with maybe a cloud with CM in it to let users know that they are client mute to client mute i mentioned this in my local mesh discord and quite a few agree that there should be some level of acknowledgement even if it's a different cloud icon in the app as it can be abit confusing