meshtastic/firmware #7330: [Feature Request]: OnScreen QWERTY keyboard for 4 way sticks

OttoTheProg, 2025-07-13:
### Platform

Cross-Platform

### Description

The Wio Tracker L1 Pro has a 4 way stick. It would be nice being able to use this for quick messages with an onscreen keyboard - like on FireTV or Android TV boxes.

Even smarter but currently no hardware available: cell phone like numeric 0-9 dialpad and at best with T9 interface. Would enable fast texting like in the old times  with minimal keys - fast and hardware efficient.

garthvh, 2025-07-13:
T9 requires a large dictionary

sim1988son, 2025-07-16:
Witam mam takie coś 

[image]

a już planuje takie 

[image]

guzik Fn/Shift zmienia mi funkcje klawiatury 

- system czyli strzałki
- alfabet duże litery
- alfabet małe litery
- cyfry