meshtastic/firmware #7355: [Bug]: Change GPS values via Remote Administration

HougeDK, 2025-07-16:
### Category

Other

### Hardware

Heltec Mesh Node T114

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [x] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.11

### Description

When trying to change fixed position lat/lon /alt on a node via Remote Administration, i get "Delivery confirmed". But nothing changes on the node

### Relevant log output

```Shell

```

mverch67, 2025-07-16:
Do you have GPS button set to enabled?

mverch67, 2025-07-16:
What do you expect to change on the node?

mverch67, 2025-07-16:
I think the may change if you press the GPS button

HougeDK, 2025-07-24:
I tried all 3 GPS modes, but it does not help. It does not change.
What is strange is that when i type in a new GPS position, i looks like it changed, when you look in the clients window. The client has the new GPS position. 
But if you reenter the GPS settings, it didn't change and efter some time it also reverts back to the old GPS position in the client window. It seems something if buffered in the APP.