meshtastic/firmware #7364: [Bug]: T114 sets incorrect date when GPS is enabled on firmware 2.7.2

AtomicJay, 2025-07-16:
### Category

Hardware Compatibility

### Hardware

Heltec Mesh Node T114

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.2.f6d3782 Alpha

### Description

When GPS is enabled, the date is set 2282 days in the past (e.g., instead of the date of writing—July 16, 2025—it shows April 17, 2019). The time of day, however, matches. When I disable GPS, the date syncs back to normal. I did not encounter such an issue on firmware 2.6.11.60ec05eb beta, which I switched back to due to this issue.

Steps to recreate:
1. Flash HelTec T114 (v2) (with GPS module) with firmware 2.7.2.f6d3782 Alpha.

2. Enable GPS.

### Relevant log output

```Shell

```

garthvh, 2025-07-16:
Probably need to wait a bit for a better lock, this is pretty common with that GPS chip

AtomicJay, 2025-07-16:
> Probably need to wait a bit for a better lock, this is pretty common with that GPS chip

I've tried leaving GPS enabled overnight with no luck.

AtomicJay, 2025-07-17:
I'll try leaving it on for a while again. Not expecting much, given my previous attempts, but I guess it's worth a try.

garthvh, 2025-07-17:
Make sure it has a clear view of the sky

jp-bennett, 2025-07-17:
Does that GPS ever get a lock? And does the date continue to be wrong if it does get a valid position?

caveman99, 2025-07-17:
The L76K is very unreliable when it comes to time caching. There should be a mechanism in the firmware where it prevents setting the date to before the time the firmware was compiled. Did you flash the node with a precompiled firmware or is this a self compiled one?

AtomicJay, 2025-07-17:
> The L76K is very unreliable when it comes to time caching. There should be a mechanism in the firmware where it prevents setting the date to before the time the firmware was compiled. Did you flash the node with a precompiled firmware or is this a self compiled one?

I flashed it with a precompiled firmware.

fifieldt, 2025-07-18:
I'm wondering whether this chunk in gps runOnce might be at play -- so even though the GPS might fix its time badness after a while, it still uses the original bad time

```    // If we've already set time from the GPS, no need to ask the GPS
    bool gotTime = (getRTCQuality() >= RTCQualityGPS);
    if (!gotTime && lookForTime()) { // Note: we count on this && short-circuiting and not resetting the RTC time
        gotTime = true;
        shouldPublish = true;
    }
```

I would expect
rtc.cpp
```RTCSetResult perhapsSetRTC(RTCQuality q, struct tm &t)```

to have a reference to the firmware build time, but that must be elsewhere.