meshtastic/firmware #7373: [Bug]: Heltec T114 will not enable MQTT

therickster-cpu, 2025-07-17:
### Category

Other

### Hardware

Heltec Mesh Node T114

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.11.60ec05e AND 2.7.2.f6d3782

### Description

No matter what I do, "mesthastic --port /dev/ttyACM0 --set mqtt.enabled true" will NOT turn on MQTT. Tried via web app, serial port and BLE (Android). Web app reports "problem when saving config" and BLE simply disconnects when you click "SEND".

This is quite annoying!

### Relevant log output

```Shell

```

garthvh, 2025-07-17:
Are you enabling the proxy as well? The t114 does not have WiFi and can not do mqtt on its own

eric-becker, 2025-07-23:
This looks like a new bug on simlar hardware (no networking). If a user forgets to enable 'Proxy to client enabled' while editing MQTT settings, none of the settings are saved and bluetooth is disabled until the node is rebooted.  Logs from my T1000-E recreating the same issue:

```
INFO  | 01:30:07 17372 Tell client we have new packets 1321
INFO  | 01:30:07 17372 BLE notify fromNum
INFO  | 01:30:07 17372 [Router] Packet History - insert: Reusing slot aged 7196.901s TRACE OLDEST SLOT
DEBUG | 01:30:07 17372 [Router] handleReceived(REMOTE) (id=0x6c19beee fr=0x00000000 to=0xcce5b101, WantAck=1, hopLim=3 Ch=0x0 Portnum=6 rxtime=1753234207 priority=70)
DEBUG | 01:30:07 17372 [Router] Module 'Admin' wantsPacket=1
INFO  | 01:30:07 17372 [Router] Received Admin from=0x0, id=0x6c19beee, portnum=6, payloadlen=92
INFO  | 01:30:07 17372 [Router] Handle admin payload 35
DEBUG | 01:30:07 17372 [Router] Client set module config
INFO  | 01:30:07 17372 [Router] Disable NRF52 bluetooth
INFO  | 01:30:07 17372 [Router] BLE Disconnected, reason = 0x16
DEBUG | 01:30:07 17372 [Router] PhoneAPI::close()
DEBUG | 01:30:07 17372 [Router] BluetoothStatus DISCONNECTED
INFO  | 01:30:07 17372 [Router] Ended BLE connection
INFO  | 01:30:07 17372 [Router] Set module config: MQTT
ERROR | 01:30:07 17372 [Router] Invalid MQTT config: proxy_to_client_enabled must be enabled on nodes that do not have a network
DEBUG | 01:30:07 17372 [Router] Partially randomized packet id 3225890155
WARN  | 01:30:07 17372 [Router] Alloc an err=32,to=0xcce5b101,idFrom=0x6c19beee,id=0xc0472d6b
DEBUG | 01:30:07 17372 [Router] Module 'Admin' considered
DEBUG | 01:30:07 17372 [Router] Discard an unneeded response
DEBUG | 01:30:07 17372 [Router] Module 'routing' wantsPacket=1
INFO  | 01:30:07 17372 [Router] Received routing from=0x0, id=0x6c19beee, portnum=6, payloadlen=92
DEBUG | 01:30:07 17372 [Router] Routing sniffing (id=0x6c19beee fr=0x00000000 to=0xcce5b101, WantAck=1, hopLim=3 Ch=0x0 Portnum=6 rxtime=1753234207 priority=70)
DEBUG | 01:30:07 17372 [Router] Partially randomized packet id 2748075372
DEBUG | 01:30:07 17372 [Router] Enqueued local (id=0xa3cc4d6c fr=0xcce5b101 to=0xcce5b101, WantAck=0, hopLim=3 Ch=0x0 Portnum=5 requestId=6c19beee rxtime=1753234207 priority=120)
DEBUG | 01:30:07 17372 [Router] Module 'routing' considered
DEBUG | 01:30:07 17372 [Router] Rx someone rebroadcasting for us (id=0xa3cc4d6c fr=0xcce5b101 to=0xcce5b101, WantAck=0, hopLim=3 Ch=0x0 Portnum=5 requestId=6c19beee rxtime=1753234207 priorit
DEBUG | 01:30:07 17372 [Router] Didn't find pending packet
INFO  | 01:30:07 17372 [Router] Packet History - insert: Reusing slot aged 7191.731s TRACE OLDEST SLOT
DEBUG | 01:30:07 17372 [Router] handleReceived(REMOTE) (id=0xa3cc4d6c fr=0xcce5b101 to=0xcce5b101, WantAck=0, hopLim=3 Ch=0x0 Portnum=5 requestId=6c19beee rxtime=1753234207 priority=120)
DEBUG | 01:30:07 17372 [Router] Module 'routing' wantsPacket=1
INFO  | 01:30:07 17372 [Router] Received routing from=0xcce5b101, id=0xa3cc4d6c, portnum=5, payloadlen=2
DEBUG | 01:30:07 17372 [Router] Routing sniffing (id=0xa3cc4d6c fr=0xcce5b101 to=0xcce5b101, WantAck=0, hopLim=3 Ch=0x0 Portnum=5 requestId=6c19beee rxtime=1753234207 priority=120)
DEBUG | 01:30:07 17372 [Router] Received a ACK for 0x6c19beee, stopping retransmissions
DEBUG | 01:30:07 17372 [Router] Delivering rx packet (id=0xa3cc4d6c fr=0xcce5b101 to=0xcce5b101, WantAck=0, hopLim=3 Ch=0x0 Portnum=5 requestId=6c19beee rxtime=1753234207 priority=120)
DEBUG | 01:30:07 17372 [Router] Update DB node 0xcce5b101, rx_time=1753234207
INFO  | 01:30:07 17372 [Router] Sort took 0 milliseconds
DEBUG | 01:30:07 17372 [Router] Forwarding to phone (id=0xa3cc4d6c fr=0xcce5b101 to=0xcce5b101, WantAck=0, hopLim=3 Ch=0x0 Portnum=5 requestId=6c19beee rxtime=1753234207 priority=120)
DEBUG | 01:30:07 17372 [Router] Module 'routing' considered
```

fifieldt, 2025-07-23:
> I was trying to get it to report its location to the SoCalMesh website, hence I was trying MQTT, under the impression that it'd send a message via LoRa to the MQTT server.

For this, you don't need to enable MQTT on the device, which is, as Garth mentions, a function that needs the internet. I had a look at the docs to see if we need an update to avoid confusion, but it does mention this this in the first line.

Instead, all you need is the toggle to tell other devices that it's OK to pass your information via MQTT. You'll find that down the bottom of MQTT settings. No need to turn on anything else.

therickster-cpu, 2025-07-23:
Ok. I don't see it at the bottom of MQTT settings in the web client. Not sure if you're referring to

```mqtt.mapReportingEnabled true```

in the CLI.

Tomorrow, I'll have a looksee with the Android app. Things appear a bit different there than the web client.

eric-becker, 2025-07-23:
It's under `LoRa` - > `OK to MQTT` enabled

PR #7446 (2025-07-26): Fix MQTT config bugs
Fixes #7373

This fixes a number of MQTT config related issues:
- The default meshtastic MQTT server now supports TLS
- Setting any module config disables BLE before the reboot. This was breaking on invalid configs and leaving devices disconnected until a reboot.
closed 2025-07-26 by thebentern