meshtastic/firmware #7374: [Bug]: Node doesn’t reboot on set canned message

RCGV1, 2025-07-18:
### Category

Other

### Hardware

Not Applicable

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7

### Description

When the client sends setCannedMessage (don’t remember exact function name) the node doesn’t reboot when it should, if rebooted then the canned messages are actually used. This issue is currently on iOS and android where setting a canned message doesn’t immediately take effect.

### Relevant log output

```Shell

```

fifieldt, 2025-07-18:
Do you see ```        LOG_INFO("Delay save of changes to disk until the open transaction is committed");
```

in the logs?

Xaositek, 2025-07-18:
Here's also a quick debrief of some testing I did this afternoon - iOS, Android, and Python CLI all have a different user experience with the same device.

**CLI**
meshtastic --set canned_message.enabled true
(Auto Reboots)
meshtastic --set-canned-message "Hi|Bye"
(Invoke a reboot by either power off / back on or through UI)
_Verify_

**Android**
Connect to node, configure Canned Message string, hit "send", no confirmation anything saved. Go over to Display and change display timeout to 90 seconds. This will cause a reboot and CannedMessages is THEN updated.

**iOS**
Connect to node, configure Canned Message string, and save, node will reboot, but the string is not saved.

jp-bennett, 2025-07-22:
Fixed with #7414 Node does not need to reboot simply to update the canned messages list.

closed 2025-08-20 by RCGV1