meshtastic/firmware #7376: [Bug]: No connect using client-test web client and HTTPS / IP address

mygooglyeyes, 2025-07-18:
### Category

WiFi

### Hardware

T-Beam

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

 2.7.3.71b6508

### Description

SSL handshake error returned by device:

Flashed my device with the absolute latest ALPHA firmware and after configuring and verifying functionality, I attempted to connect to my device using the client-test web client (not self hosted) using HTTPS.  

open client-test.meshtastic.org
click the new connection button
select HTTPS
enter the IP address of the device and click connect
<yellow error message is returned in client-test window> 
right click on link in error message and open in new tab
click the "advanced" option on the warning page to continue to connect
click the option to connect
Connection is reset by the client web page is shown, and no connection can be made no matter how many more times I try

I can connect via the Android app using HTTPS.

I have a .json log of the event captured by chrome if anyone wants to look at it - I can't decode it myself.

[chrome-net-export-log-2.json](https://github.com/user-attachments/files/21309249/chrome-net-export-log-2.json)

### Relevant log output

```Shell
Serial output of the client:
 
[HTTPS:I] New connection. Socket FID=53
[HTTPS:E] SSL handshake failed. Aborting handshake. FID=53
[HTTPS:I] Connection closed. Socket FID=53
[HTTPS:I] New connection. Socket FID=53
[HTTPS:E] SSL handshake failed. Aborting handshake. FID=53
[HTTPS:I] Connection closed. Socket FID=53
[HTTPS:I] New connection. Socket FID=53
[HTTPS:E] SSL handshake failed. Aborting handshake. FID=53
[HTTPS:I] Connection closed. Socket FID=53
```

mrbaranovskiy, 2026-01-29:
Not planned? Because of this issue, Meshtastic is quite pointless. I have no serial cable on the roof...

fifieldt, 2026-04-05:
> I feel that esp32 WiFi is not a priority of meshtatsic developers see [#8827](https://github.com/meshtastic/firmware/issues/8827)

I am sorry you feel this. To be honest, we're just a bit overwhelmed and need help :(

miky2k, 2026-09-02:
> > I feel that esp32 WiFi is not a priority of meshtatsic developers see [#8827](https://github.com/meshtastic/firmware/issues/8827)
> 
> I am sorry you feel this. To be honest, we're just a bit overwhelmed and need help :(

No AI ? Do you need donation to buy AI plan ?

jp-bennett, 2026-10-06:
Probably fixed by 11838

closed 2026-10-06 by jp-bennett