meshtastic/firmware #7377: [Bug]: heltec wireless paper - Restarts

SQ9PXB, 2025-07-18:
### Category

Other

### Hardware

Heltec Wireless Paper

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [x] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.2

### Description

The device restarts frequently.

### Relevant log output

```Shell
[0m[34mDEBUG [0m| 02:22:09 1382 [Renderer] [34mExplicit FAST
[0m[32mINFO  [0m| 02:22:09 1382 [Renderer] [32mUpdating display
[0m[34mDEBUG [0m| 02:22:09 1383 [TwoButton] [34mState: ON
[0m[34mDEBUG [0m| 02:22:10 1383 [Renderer] [34mFULL-update debt:15.599983
[0m[34mDEBUG [0m| 02:22:10 1383 [Renderer] [34mExplicit FAST
[0m[32mINFO  [0m| 02:22:10 1383 [Renderer] [32mUpdating display
[0m[34mDEBUG [0m| 02:22:10 1384 [TwoButton] [34mState: ON
[0m[34mDEBUG [0m| 02:22:11 1384 [Renderer] [34mFULL-update debt:15.799983
[0m[34mDEBUG [0m| 02:22:11 1384 [Renderer] [34mExplicit FAST
[0m[32mINFO  [0m| 02:22:11 1384 [Renderer] [32mUpdating display
[0m[34mDEBUG [0m| 02:22:12 1385 [Power] [34mBattery: usbPower=0, isCharging=0, batMv=3908, batPct=71
[0m[34mDEBUG [0m| 02:22:12 1386 [TwoButton] [34mState: ON
[0m[34mDEBUG [0m| 02:22:13 1386 [Renderer] [34mFULL-update debt:15.999983
[0m[34mDEBUG [0m| 02:22:13 1386 [Renderer] [34mExplicit FAST
[0m
assert failed: void NicheGraphics::InkHUD::Applet::render() Applet.cpp:60 (assignedTile)

Backtrace: 0x40377b42:0x3fcebd30 0x40380fb9:0x3fcebd50 0x40387719:0x3fcebd70 0x42074bd5:0x3fcebea0 0x42065c31:0x3fcebec0 0x42065f99:0x3fcebf10 0x42079ffa:0x3fcebf30 0x420617ed:0x3fcebf50 0x4201a411:0x3fcebf90

ELF file SHA256: ddfffd9a3c676734

Rebooting...
 �ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42148a66
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd

[0m[34mDEBUG [0m| 02:45:41 495 [Router] [34mModule 'HeardApplet' considered
[0m[34mDEBUG [0m| 02:45:41 495 [Renderer] [34mFULL-update debt:0.600000
[0m[34mDEBUG [0m| 02:45:41 495 [Renderer] [34mUNSPECIFIED: using FAST
[0m[34mDEBUG [0m| 02:45:41 495 [Renderer] [34mChannel 0 took 17ms to render
[0mGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42071261  PS      : 0x00060030  A0      : 0x82074c33  A1      : 0x3fcebd50  
A2      : 0x3fcb1888  A3      : 0x0000006e  A4      : 0x00000000  A5      : 0x000000ff  
A6      : 0x0000007a  A7      : 0x0000005f  A8      : 0x00000003  A9      : 0x3fcebd30  
A10     : 0x00000118  A11     : 0x0000007a  A12     : 0x0000006e  A13     : 0x3fcebe68  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000bb  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x4207125e:0x3fcebd50 0x42074c30:0x3fcebea0 0x42065c31:0x3fcebec0 0x42065f99:0x3fcebf10 0x42079ffa:0x3fcebf30 0x420617ed:0x3fcebf50 0x4201a411:0x3fcebf90

ELF file SHA256: ddfffd9a3c676734

Rebooting...
 �ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42148a66
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc7
```

todd-herbert, 2025-07-18:
Have you got any more information about what's triggering this? The first crash looks like it was while spamming the button, right? What's the context for the second crash?

todd-herbert, 2025-07-18:
I've got a suspicion about the first one, but not sure about the second one at all. I haven't seen it come up here in testing and can't decode any information from that exception for some reason. It could be worth trying a full erase, I've seen it resolve weird things like this before.