meshtastic/firmware #7384: [Bug]: Linux native meshtasticd SEGV

brauhausdc, 2025-07-19:
### Category

Other

### Hardware

Linux Native

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.11.25

### Description

Linux Native meshtasticd has been segv’ing when I try to start it since installing 2.6.11.25:

hostname:~$ sudo systemctl status meshtasticd 
* meshtasticd.service - Meshtastic Native Daemon
     Loaded: loaded (/lib/systemd/system/meshtasticd.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Sat 2025-07-19 11:57:36 GMT; 2s ago
    Process: 26676 ExecStart=/usr/bin/meshtasticd (code=killed, signal=SEGV)
   Main PID: 26676 (code=killed, signal=SEGV)
        CPU: 114ms
 

hostname:~$ apt list --installed mesh\* 
Listing... Done
meshtasticd/unknown,now 2.6.11.25~obs60ec05e~beta armhf [installed]

Using a MeshAdv board on a Pi-2b 1.1 with a Reasonably current version of RaspPi OS. Worked fine before. 

### Relevant log output

```Shell

```

brauhausdc, 2025-07-20:
> What happens if you run manually /usr/bin/meshtasticd in a terminal window? Post the output under Relevant log output above.

malloc/free issue maybe? 

***@***.***:~$ sudo /usr/bin/meshtasticd 
Portduino is starting, VFS root at /root/.portduino/default
Set up Meshtastic on Portduino...
free(): invalid pointer
Aborted

Or as root: 

***@***.***:~# sudo /usr/bin/meshtasticd
Portduino is starting, VFS root at /root/.portduino/default
Set up Meshtastic on Portduino...
Segmentation fault

Trying it with gdb (but mostly failing, since it wasn’t compiled with debugging symbols) 

***@***.***:~# gdb /usr/bin/meshtasticd 
GNU gdb (Raspbian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/meshtasticd...
(No debugging symbols found in /usr/bin/meshtasticd)
(gdb) run
Starting program: /usr/bin/meshtasticd  [Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Portduino is starting, VFS root at /root/.portduino/default
Set up Meshtastic on Portduino...

Program received signal SIGSEGV, Segmentation fault.
0x0046916e in ?? ()

mverch67, 2025-07-20:
> Program received signal SIGSEGV, Segmentation fault.
> 0x0046916e in ?? ()

you forgot to type `where` or `bt` to see the full stacktrace.

brauhausdc, 2025-08-06:
Any updates here? Is this on anyone's radar to address?

ford-jones, 2025-08-13:
Have you tried taking a look at this in valgrind?

konimaru, 2025-08-15:
Another option would be `$ strace /usr/bin/meshtasticd` (if available).

konimaru, 2025-08-15:
> Correct me if I’m wrong here, but unless the binary is built with debugging turned on (which the distributed one isn’t) any of these efforts are going to be largely fruitless. I’d humbly ask if this part of the project has been deprecated — and if it hasn’t, suggest that it should be since it has apparently been abandoned by the developer. --Dan BrownOn Aug 15, 2025, at 04:32
```
#include <stdio.h>

int main(int argc, char **argv)
{
    printf("Hello world!\n");
    return 0;
}
```
```
$ gcc +O3 sample.c -o sample
$ strip sample
```
This leaves us without debug info and no symbols.
```
$ strace ./sample
execve("./dummy", ["./dummy"], 0x7fff078cadb0 /* 65 vars */) = 0
brk(NULL)                               = 0x651da4fb4000
arch_prctl(0x3001 /* ARCH_??? */, 0x7fffc4547b10) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7a42a157a000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=91572, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 91572, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7a42a1563000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P\237\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0 \0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0"..., 48, 848) = 48
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\325\31p\226\367\t\200\30)\261\30\257\33|\366c"..., 68, 896) = 68
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=2220400, ...}, AT_EMPTY_PATH) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
mmap(NULL, 2264656, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7a42a1200000
mprotect(0x7a42a1228000, 2023424, PROT_NONE) = 0
mmap(0x7a42a1228000, 1658880, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x28000) = 0x7a42a1228000
mmap(0x7a42a13bd000, 360448, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1bd000) = 0x7a42a13bd000
mmap(0x7a42a1416000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x215000) = 0x7a42a1416000
mmap(0x7a42a141c000, 52816, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7a42a141c000
close(3)                                = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7a42a1560000
arch_prctl(ARCH_SET_FS, 0x7a42a1560740) = 0
set_tid_address(0x7a42a1560a10)         = 11308
set_robust_list(0x7a42a1560a20, 24)     = 0
rseq(0x7a42a15610e0, 0x20, 0, 0x53053053) = 0
mprotect(0x7a42a1416000, 16384, PROT_READ) = 0
mprotect(0x651d9c78c000, 4096, PROT_READ) = 0
mprotect(0x7a42a15b4000, 8192, PROT_READ) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
munmap(0x7a42a1563000, 91572)           = 0
newfstatat(1, "", {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0x3), ...}, AT_EMPTY_PATH) = 0
getrandom("\x67\xff\x26\xd7\xb7\x7b\xcf\xbb", 8, GRND_NONBLOCK) = 8
brk(NULL)                               = 0x651da4fb4000
brk(0x651da4fd5000)                     = 0x651da4fd5000
write(1, "Hello world!\n", 13Hello world!
)          = 13
exit_group(0)                           = ?
+++ exited with 0 +++
```
That in turn should be plenty to at least get an idea where it goes wrong.

brauhausdc, 2025-08-24:
Any other debugging you all need? Is this getting any attention?

brauhausdc, 2025-08-24:
> Have you defined MaxNodes: 200 under General: in the config.yaml? If not it may crash.

Yes, MaxNodes: 200 is defined. 

General:
  MaxNodes: 200
  MaxMessageQueue: 100

mverch67, 2025-08-24:
Are you able to [compile](https://meshtastic.org/docs/development/linux/#building) the meshtasticd code by yourself (with debug info)?

`pio run -e native-tft`

brauhausdc, 2025-08-24:
> Seems there's a bug or inconsistency in your config.yaml that is not handled properly by the yaml library.
> I suggest you use the original config.yaml for your meshtasticd version and check if it works. Then, make your changes incrementally step by step.

Ok - 

So I — 

Systemctl stopped and disabled meshtasticd 

Apt purge’ed meshtasticd and meshtasticd-dbgsym. 

Mv’ed /etc/meshtasticd to /etc/meshtasticd_not 

Rm’ed /root/.portduino 

Then, apt installed meshtasticd-dbgsym which also installed meshtasticd 

Checked to see if config.yaml had been installed… and… it was not. In previous revisions, it had been. 

***@***.***:/etc/meshtasticd# find . 
.
./config.d
./ssl
./available.d
./available.d/femtofox
./available.d/MUI
./available.d/OpenWRT

I checked systemctl status meshtasticd, and Meshtastic appears to run, but in simulation mode, when there is no config file. 

I did a “systemctl stop meshtasticd” 

Then I downloaded the config file you pointed to, put it into /etc/meshtasticd/config.yaml. I made Zero changes. Nothing. Nada. 

pi:~$ sudo strace -o messtastic.out  `which meshtasticd `    
Portduino is starting, VFS root at /root/.portduino/default
Set up Meshtastic on Portduino...
Segmentation fault

My conclusion — This is NOT a problem with anything I’ve put into the config file. There is a bug with Meshtasticd centered around reading its config file. I would kindly ask the developers to have a look. 

I’d also suggest that adding the config files information back into the Debian package would be a good idea, since they appear to have been removed (see “find” above).

closed 2025-08-24 by mverch67