meshtastic/firmware #7431: [Bug]: nrf52840 repeatedly resets when using the serial module's PROTO mode.

Quency-D, 2025-07-23:
### Category

Serial

### Hardware

Not Applicable

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.4

### Description

After configuring the serial port module to the configuration shown below, it cannot run normally and keeps resetting.
```json
  "serial": {
    "enabled": true,
    "baud": "BAUD_115200",
    "timeout": 300,
    "mode": "PROTO",
    "overrideConsoleSerialPort": true,
    "echo": false,
    "rxd": 0,
    "txd": 0
  }
```

When trying to find the cause, I found that commenting out the following statements would prevent the reset, but I still don't know what these statements do.

```c++
void StreamAPI::emitTxBuffer(size_t len)
{
    if (len != 0) {
        txBuf[0] = START1;
        txBuf[1] = START2;
        txBuf[2] = (len >> 8) & 0xff;
        txBuf[3] = len & 0xff;

        auto totalLen = len + HEADER_LEN;
        // stream->write(txBuf, totalLen);
        // stream->flush();
    }
}
```

### Relevant log output

```Shell
//\ E S H T /\ S T / C

[0m[32mINFO  [0m| ??:??:?? 0 [32mWait for peripherals to stabilize
[0m[34mDEBUG [0m| ??:??:?? 1 [34mFilesystem files:
[0m[34mDEBUG [0m| ??:??:?? 1 [34m adafruit (directory)
[0m[34mDEBUG [0m| ??:??:?? 1 [34m bond_prph (directory)
[0m[34mDEBUG [0m| ??:??:?? 1 [34m bond_cntr (directory)
[0m[34mDEBUG [0m| ??:??:?? 1 [34m prefs (directory)
[0m[34mDEBUG [0m| ??:??:?? 1 [34m   channels.proto (57 Bytes)
[0m[34mDEBUG [0m| ??:??:?? 1 [34m   config.proto (166 Bytes)
[0m[34mDEBUG [0m| ??:??:?? 1 [34m   device.proto (137 Bytes)
[0m[34mDEBUG [0m| ??:??:?? 1 [34m   module.proto (109 Bytes)
[0m[34mDEBUG [0m| ??:??:?? 1 [34m   nodes.proto (1450 Bytes)
[0m[34mDEBUG [0m| ??:??:?? 1 [34mPower::lipoInit lipo sensor is not ready yet
[0m[34mDEBUG [0m| ??:??:?? 1 [34mUse analog input 4 for battery level
[0m[32mINFO  [0m| ??:??:?? 1 [32mScan for i2c devices
[0m[34mDEBUG [0m| ??:??:?? 1 [34mScan for I2C devices on port 2
[0m[34mDEBUG [0m| ??:??:?? 1 [34mScan for I2C devices on port 1
[0m[32mINFO  [0m| ??:??:?? 1 [32mNo I2C devices found
[0m[34mDEBUG [0m| ??:??:?? 1 [34macc_info = 0
[0m[32mINFO  [0m| ??:??:?? 1 [32mS:B:69,2.7.4.ed0cdefb4,heltec-mesh-node-t114,git@github.com:meshtastic/firmware
[0m[34mDEBUG [0m| ??:??:?? 1 [34mReset reason: 0x0
[0m[34mDEBUG [0m| ??:??:?? 1 [34mSet random seed 3291078315
[0m[32mINFO  [0m| ??:??:?? 1 [32mInit NodeDB
[0m[32mINFO  [0m| ??:??:?? 1 [32mLoad /prefs/nodes.proto
[0m[32mINFO  [0m| ??:??:?? 1 [32mLoaded /prefs/nodes.proto successfully
[0m[32mINFO  [0m| ??:??:?? 1 [32mLoaded saved nodedatabase version 24, with nodes count: 14
```

mverch67, 2025-07-23:
> "rxd": 0,
    "txd": 0

looks wrong. rx/tx cannot be the same GPIO port

Quency-D, 2025-07-24:
I have the same problem with 2.7.3.cf574c7 Alpha firmware.

mverch67, 2025-07-24:
Maybe then you need to switch off the debug log?

thebentern, 2025-07-24:
Hi @Quency-D, `overrideConsoleSerialPort: true` causes the serial module to work over USB, which does not work in the PROTO mode, since it already is handling the serial logging or another proto interface. This only works with NMEA and CALTOPO modes which output nmea sentences in addition to the serial logs.

PR #7470 (2025-07-27): Validate Serial config console override modes
Fixes #7431 by validating the serial mode config so that we don't put devices into a bad state / boot loop