meshtastic/firmware #7452: [Feature Request]: Do node info exchange if we rx a DM from a node we do not have a pub key for

RCGV1, 2025-07-25:
### Platform

Cross-Platform

### Description

I have verified that a node with a full db will not attempt to do a node info exchange if it receives a dm from a node that is unknown to it. This results in a "unknown pub key" error and could easily be automatically fixed with a node info exchange between the nodes. This happens when one node has another in its db, but that other node doesn't have the sender in its db.

GUVWAF, 2025-07-25:
This should be already there, but first the routing error is sent on the primary channel (in the hope both have that channel):
https://github.com/meshtastic/firmware/blob/bbe548bc980827145114b76260b1b2377dd633e5/src/mesh/ReliableRouter.cpp#L107-L108
When receiving this routing error, the other side sends its NodeInfo:
https://github.com/meshtastic/firmware/blob/bbe548bc980827145114b76260b1b2377dd633e5/src/mesh/ReliableRouter.cpp#L119-L120

It might be it doesn't send the NodeInfo when it already did so within the last 1 min., but then the other side should have its public key already.

hotellonely, 2025-09-07:
Currently the worst case is when a node has lost its previous key. Then essentially it would need to be deleted from EVERY OTHER DEVICE it has ever exchanged node info with.