meshtastic/firmware #7466: [Bug] GPS is turned off too soon after lock, causing problems

fifieldt, 2025-07-26:
As investigated by @Quency-D , we turn off GPS more or less instantly after getting lock.

However, GPS chips are designed to stay locked for a while to download some data and save it. This data is important for speeding up future locks, and making them higher quality.

We started work in https://github.com/meshtastic/firmware/pull/7234 , but this was Quency's master branch, so we need a new place to work. Creating this issue so the problem is not lost and will be fixed.

Generally, to fix this we should consider making a new state in our GPS FSM for Lock, and pausing there for a while.

As for how much to pause - at least 20 seconds, maybe up to 60 seconds. This may change depending on first lock or subsequent locks.

Related https://github.com/meshtastic/firmware/pull/7234

b8b8, 2025-08-06:
[image]

RichieFrame, 2025-08-31:
I tried 15 and 10s intervals based on the above image, 15s helped but 10s brought the precision way higher, before it would jump around +- 25 meters or so in random directions, now its within 2 meters

b8b8, 2025-08-31:
> I tried 15 and 10s intervals based on the above image, 15s helped but 10s brought the precision way higher, before it would jump around +- 25 meters or so in random directions, now its within 2 meters

Everything 10 seconds and under keeps the GPS always on, that's why I chose 8 seconds.

closed 2025-09-02 by thebentern