meshtastic/firmware #7469: [Feature Request]: Telemetry (indoor or outdoor) environment

livingdeath72, 2025-07-26:
### Platform

other

### Description

Could an option be added to the telemetry module in Meshtastic to indicate whether environmental data is being measured indoors or outdoors?

[image]

[image]

livingdeath72, 2025-07-28:
> Would you be able to add a bit of context? e.g. not what you want changed but what you want to accomplish with the change?

It is your choice whether the app measures environmental values inside or outside. Now you can only guess where the measurements were taken.

This information would then be visible to everyone in the app.

ftrueck, 2025-08-11:
I would go a step further and transform this into a field called data source that can be a short text (e.g. 10 chars). Because there is not only indoor and outdoor: There is green houses, factory, machine 1, etc.

ftrueck, 2025-08-11:
> > I would go a step further and transform this into a field called data source that can be a short text (e.g. 10 chars). Because there is not only indoor and outdoor: There is green houses, factory, machine 1, etc. If a text is too much traffic I suggest to use a number field.
> 
> Good idea, and the different measurement use cases could also be selectable from a drop-down menu?

A dropdown would imply everyone has the same use cases. That means you have to define every use case in advance and everyone has to agree to that. I rather prefer a more open approach where the admin of the node defines what the purpose of the data is.