meshtastic/firmware #7478: [Bug]: Fixed GPS coordinates altered on save

zomgugoff, 2025-07-27:
### Category

Other

### Hardware

T-Beam S3

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.3

### Description

Saving coordinates for fixed position use is resulting in some sort of math(rounding?) that is chaning the numbers and mis-represeting the location.

`--setlat 46.57574`

This number is being stored as 46.575739999999996, although this changes depending on the client (CLI vs Android app)

But a 3-digit whole number prevents this adjustment:

`--setlat 146.57574`

### Relevant log output

```Shell

```

moverhorst-pitcrew, 2025-08-12:
It most likely has to do with the precision. On my T-beam i had the same issue, setting precision through meshtastic cli to 32(?) i think make it accurate and persistent.

ford-jones, 2025-08-17:
I was unable to reproduce this on the heltec v3 using a fixed position. The mesh in my area is fairly quiet though, it may be that I never received the coordinates I was sending out back again.

heeplr, 2025-10-17:
I experience the same issue on a T-Deck with 2.6.11 ... I've set LongFast to "Precise location" (yellow lock turned to red lock) and now it seems the position stays unaltered.

I think this is a bug since we want the "rounded" coarse precision configured only for some channels that don't have precise location, not globally for all channels.

gizmo69the2nd, 2026-02-22:
Issue still present in 2.7.19.bb3d6d5 on RAK4631.
Suggestion from [ViezeVingertjes](https://github.com/ViezeVingertjes) and [heeplr](https://github.com/heeplr) "fixed" it, but not ideal.

> I experience the same issue on a T-Deck with 2.6.11 ... I've set LongFast to "Precise location" (yellow lock turned to red lock) and now it seems the position stays unaltered.
>

closed 2026-08-31 by github-actions[bot]