meshtastic/firmware #7481: [Feature Request]: 3.x have narrow bandwidth and low spreading factor presets

mikecarper, 2025-07-27:
### Platform

Cross-Platform

### Description

Using a narrow bandwidth like 62.5 and lower allows for a lower noise floor to be found. Add in some presets that support 62.5khz SF7-9; and then for sx 1262 chips support 8khz SF5 up to 41.7khz SF5-8. Basically limit the lowest Kbps to be bigger than 0.8kbps with the new presets.

mikecarper, 2025-08-10:
[image]

closed 2025-10-05 by github-actions[bot]

geirgp, 2025-11-28:
Norwegian community has due to interference on  869.525 been testing these settings over the last couple of weeks: 
869.618Mhz 
62.5khz 
SF8 
CR 4/5 

This has given us much better reliability and range than stock LongFast in those noisy areas. Reliability (delivery, acks) increased significantly, as did range.

Right now this all requires manual custom LoRa settings, which is a lot to ask from new users,
--> We need a Narrow preset.

mikecarper, 2026-01-04:
[image]

Here's where the names land. If Narrow Moderate is used and not Narrow Medium, that puts them outside of slot 4. Narrow Slow would be using SF9/SF10 most likely so a different SF value isn't as catastrophic as using the same SF value in terms of interference.

garthvh, 2026-01-04:
These would be European only presets.