meshtastic/firmware #7496: [Bug]: WisMesh Ethernet MQTT Gateway reporting noEthernet = true

netureza-pt, 2025-07-29:
### What happened?

Under "Radio Configuration" there is no "Network" option to configure WiFi or Ethernet.

Is it supposed to only show when the device has WiFi or Ethernet hardware?

My device (WisMesh Ethernet MQTT Gateway) has ethernet. When powered by USB or PoE the ethernet link is established at 100Mb/s full duplex, but there is no dhcp request and no way to configure it in the app.

I am not sure this is a problem in the app, the firmware or the hardware.

See also my [post on RAKWireless forum](https://forum.rakwireless.com/t/no-network-on-wismesh-ethernet-mqtt-gateway/15245/2)

### App Version

2.6.30

### Phone

Asus Zenphone
LG Q6

### Device

WisMesh Ethernet MQTT Gateway (with PoE)

### Firmware

2.6.11.60ec05e

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

cromelex, 2025-07-29:
I'm not sure if this is somehow addressed in one of the alpha firmwares, but this is working for me on firmware V2.7.3 with Android app 2.6.30.

I did see some weirdness when I first plugged it in with the original firmware, as if dhcp wasn't working.

Once I flashed the newer firmware, it worked without issue. 

So whatever caused this might actually have already been fixed on the latest firmware.

netureza-pt, 2025-07-30:
I flashed 2.7.3 and it's working! Many thanks for your input!

closed 2025-07-30 by netureza-pt