meshtastic/firmware #7534: [Bug]: Heltec v3.2 - Display is empty

Tommy-01, 2025-08-02:
### Category

Other

### Hardware

Heltec V3

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [x] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.11.60ec05e

### Description

Hello,
i flashed my new Heltec v3.2 with the WebFlasher.
It works but the Display stays dark.

Cann you please help?

### Relevant log output

```Shell

```

mverch67, 2025-08-02:
Maybe the display is broken? How do you know it's ok?

Tommy-01, 2025-08-02:
The FAQs from Heltec says, there is an Display Issue with Meshtastic.

https://heltec.org/project/wifi-lora-32-v3/

`Meshtastic screen has a digital front-end issue?

It’s caused by the meshtastic setting. Changing the default screen to SSD1306 can solve the problem.

The specific steps are as follows:

“Radio configuration”→”Display”→”Override OLED auto-detect”→”SSD1306”`

Not this did not work for me.
Maybe this hint is not for v3.2

Tommy-01, 2025-08-06:
The Display is broken. Sry, i didn't see it right away.

closed 2025-08-06 by Tommy-01

SnkMtn000, 2026-03-02:
I just got a brandy new Heltec V3 (v3.2) same issue using arduino 2.x IDE and #include <heltec_unofficial.h> under the examples provided by the Heltec library, BUT using their board definition.  Yes the display is very DIM, barely viewable.  It is NOT a display or board issue as running another demo program, SSD1306SimpleDemo.ino, #include "SSD1306Wire.h"  // legacy: #include "SSD1306.h", and it is bright as bright can be.   

Just thought I would share before anyone discards a perfectly functional ESP32 LORA OLED board.

:)