meshtastic/firmware #7539: [Bug]: Heltec V2

klausz, 2025-08-03:
### Category

Other

### Hardware

Heltec v2

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.2.14

### Description

It seems WiFi LoRa 32(V2), ESP32 + SX1262 LoRa Node, Meshtastic and LoRaWAN compatible — Phaseout
has gone. 
There is no firmware. 
Why dont you tell? 

### Relevant log output

```Shell

```

closed 2025-08-03 by garthvh

klausz, 2025-08-03:
Decision done. So there was no information on excluding this hardware published. 
I had two of them. So I throw them into the bin. 
Meshtastic will not be used in the upcoming future. Too expensive.

b8b8, 2025-08-03:
The heltec Lora32 is wayy better or step it up to the Heltec T114 and get 10 times the battery since its NRF based, not ESP32.