meshtastic/firmware #7553: [Feature Request]: RGB LED for node status

CypressXt, 2025-08-05:
### Platform

Cross-Platform

### Description

Hello there 👋, 

For devices that support RGB, would you be interested in being able to use the LED as a device's status indication?
As a first implementation I thought about those status: 
* Charging (breathing blue LED)
* Low battery (blinking red LED)
* Lost GPS lock (Blinking orange LED)
* Default color set by the ambient lighting module
* Temporary override the LED status by the external notification module when needed

This would be in addition of the current ambient lighting and external notification module features and you should be able to use the LED status in addition of the 2 already existing modules.

Let me know what you think, if that make sense to everyone I can gladly prepare a PR with it 🚀

jp-bennett, 2025-08-05:
I have a NeoPixel hanging off my Raspberry Pi, and I've been thinking about something very similar. Watching this.

melkati, 2026-01-07:
Hello, is there any progress on this? 
I would like to use the onboard LED of the ESP32 S3 Supermini.

CypressXt, 2026-04-21:
Awesome, thanks for the additional context !
Let's see what we can achieve there 🚀

CypressXt, 2026-05-02:
At the current stage of the work, to be able to use the RGB device status module your LED need to be supported by the current [Ambient Lighting](https://github.com/meshtastic/firmware/blob/develop/src/AmbientLightingThread.h) module.

Or you can probably map headers as done here: [ELECROW-ThinkNode-M5/variant.h#L9-L14](https://github.com/meshtastic/firmware/blob/develop/variants/esp32s3/ELECROW-ThinkNode-M5/variant.h#L9-L14)