meshtastic/firmware #7572: [Feature Request]: Prefer message with highest hop_limit when forwarding.

foch, 2025-08-07:
### Platform

Cross-Platform

### Description

### Overview
This feature aims to improve message propagation reliability by **prioritizing messages with the highest `hop_limit`**. The goal is to ensure that all reachable nodes within the allowed hop range are contacted — regardless of poor infrastructure (e.g., misconfigured routers) or timing issues (e.g., race conditions).

### Rationale
When a message takes multiple paths to reach a node, the version with the **highest `hop_limit`** typically came through the **shortest path**. Favoring this version:

- Increases the chances of forwarding to more distant nodes for the same `hop_limit`.
- Mitigates the impact of hop-consuming infrastructure.
- Reduces the influence of luck on message delivery.

### Implementation Strategy

This logic can be split into 2 use cases:

#### 1. Message Still in Forwarding Queue
If a message has been received and is **waiting in the forwarding queue**, and another copy arrives via a different path with a **higher `hop_limit`**, the newer version should **replace** the one in the queue.

#### 2. Message Already Forwarded
If a message has already been forwarded, and a new copy arrives with a **higher `hop_limit`**, it should be **forwarded again**.

> ✅ This gives nodes beyond the previous reach another chance to receive the message.

### 📈 Diagram
The second use case is illustrated below for clarification:

[image]

ftrueck, 2025-08-11:
I see problems in use case #2.
This assumes there is only one path of transmitting where the node is a key role in the network. 
If there are multiple paths this strategy leads to high network traffic reducing overall network performance and bandwidth and increases drastically the air time.
The danger is that this strategy is flooding the network within short amount of time reducing overall effectivity even more that it could be of use to.
Since the nodes are sending omni-directional, due to the nature of radio signals, there is no way to prevent backfire and forward flodding. So instead of finding the most effective path the messages can go backwards towards the original sender.
I think this strategy will not be of benefit for the network.

ftrueck, 2025-08-11:
Forgot to mention: use case 1 seems legit to me. If the devs agree that this is not harming the mesh I see no problems that someone could implement it.

foch, 2025-08-11:
Thanks for the feedback, I will happily give a try at step 1 after my holidays unless blocking feedback is raised before or someone more qualified than me implements it before.

closed 2025-10-05 by GUVWAF