meshtastic/firmware #7575: [Bug]: Large volume of messages over BLE causes node to disconnect and require restart

michaelmedellin, 2025-08-07:
### Category

BLE

### Hardware

Station G2

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.11.60ec05e

### Description

I'm conducting some local testing with a simple, custom iOS app that connects to a node over BLE and sends ~100 messages. After I disconnect the radio from bluetooth, reconnect, and conduct the same test of 100 messages (no ack, no hops, only adjacent nodes) the radio apparently crashes because it disconnects from the iOS device and I must restart the node to reconnect. If I do not restart the node I'm not able to reconnect to the node. Sometimes I'm able to continue with the test successfully and other times the radio will crash again, this seems to be non-deterministic. 

If I do a factory reset of the node before trying the test again, it does work successfully. 

My code sends messages serially, waiting for the Core Bluetooth `didWriteValueFor` delegate to be called for each message written to the `TORADIO` characteristic, code below. I do not introduce any other delay before continuing with the write to the characteristic. 

Test Message app data encoded and sent:
```swift 
do {
     let data = try toRadio.serializedData()
            
     return await withCheckedContinuation { continuation in
          self.pendingWriteConfirmation = continuation
          peripheral.writeValue(data, for: characteristic, type: .withResponse)
          print("📡 Sent message ID: \(meshPacket.id), awaiting write confirmation...")
         }
      } catch {
     print("Failed to send message: \(error)")
    return false
}
```

And the delegate callback, that allows the continuation to continue processing:
```swift
func peripheral(_ peripheral: CBPeripheral, didWriteValueFor characteristic: CBCharacteristic, error: Error?) {
        if let error = error {
            print(" Write failed: \(error.localizedDescription)")
            pendingWriteConfirmation?.resume(returning: false)
        } else {
            print("\Write successful")
            pendingWriteConfirmation?.resume(returning: true)
        }
        pendingWriteConfirmation = nil
    }
```

While I realize Meshtastic is not specifically designed for these types of loads, I'm trying to understand where the failure is introduced to improve the reliability of the bluetooth connection. It would be nice for the radios to fail the write to the characteristic if there's a load issue, otherwise a successful write should be handled successfully on the radio. 

I've also reproduced this behavior using my Heltec V3 nodes, so I do not believe it's limited to a specific board.

Happy to take any direction for further testing to help narrow down a cause, or provide my basic code for others to test. Really appreciate any guidance.

### Relevant log output

```Shell

```

garthvh, 2025-08-07:
Is this with power saving on turning off the bluetooth again?

michaelmedellin, 2025-08-08:
Thank you so much for the speedy reply. Setting `powerConfig.isPowerSaving = false` based on your recommendation and still seeing the issue where the radio will disconnect during the send loop. 

Logging messages `FROMRADIO`: 
```
config {
  power {
    ls_secs: 300
  }
}
```

I'm not completely familiar with the Meshtastic power saving configuration so let me know if I've missed something. I'm getting to about 5 messages before the radio disconnects. I went ahead and also re-flashed the firmware with a full erase as well. 

If this is expected behavior and documented, happy to review. It just seems really difficult to pin down consistent behavior: just now I pressed the reset button on the radio and it came back up without sending the full config after a wantConfig protobuf was written TORADIO. Restarting the radio once more seems to fixed that, but again seems difficult to reproduce. 

Again really appreciate the guidance and info. 

Cheers.

closed 2025-08-19 by michaelmedellin