meshtastic/firmware #7594: [Bug]: mqtt cant enable TLS

shortwavesurfer2009, 2025-08-10:
### Category

Other

### Hardware

Seeed Card Tracker T1000-E

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.4

### Description

I just installed 2.7.4 and when i attempt to enable TLS on the default mqtt server i get

"Invalid MQTT config: default server address must not have a port specified"

The server has no port, just the url

### Relevant log output

```Shell

```

garthvh, 2025-08-10:
There is no WiFi on the tracker card, so it would have to use the mqtt proxy for mqtt, are you using android?

shortwavesurfer2009, 2025-08-10:
Yes, I am using Android and I have proxy2 client enabled in order to transfer data to the internet and from the internet using the phone's data connection.

closed 2025-08-12 by garthvh

shortwavesurfer2009, 2025-08-12:
> you need to upgrade your firmware to 2.7.3 to enable TLS on the public server via the proxy

[image]