meshtastic/firmware #7597: [Feature Request]: Enhanced Message Delivery Status - Show Receiving Node Details

Fabryz, 2025-08-10:
### Contact Details

.fabryz

### Tell us your idea.

Currently, when sending a message from the app, a cloud icon appears on the right side of the message. If the message is not received by anyone, the cloud gets crossed out after some time. If someone receives it, a checkmark appears, and tapping on it shows an overlay saying "Message delivery status: Acknowledged".

**The Problem:**
In areas with sparse node coverage and transmission issues, users need to test and find optimal transmission points. The current "Acknowledged" status only confirms that someone received the message, but doesn't provide crucial information about who received it. This creates uncertainty about whether the acknowledgment came from the user's own nodes (even when set to CLIENT_MUTE) or from other network participants.

**Proposed Enhancement:**
Enhance the message delivery status overlay to include detailed information about the receiving node(s):

- Node Name/ID: Display the name or identifier of the first node that acknowledged the message
- Node Verification: Allow users to verify if the acknowledging node is one of their own devices or an external node
  - Additional Node Information: Show relevant details about the receiving node such as:
  - Distance/location information (if available)
  - Signal strength/SNR
  - Node position on map (if location data is available)
  - Timestamp of acknowledgment

**Use Case Benefits:**

- Coverage Testing: Users can determine the effective range and coverage of their transmissions
- Network Verification: Confirm successful communication with external nodes vs. own devices
- Optimization: Identify optimal transmission locations based on which nodes are reachable
- Network Mapping: Better understand the local mesh network topology and node distribution

**Suggested Implementation:**
Instead of just showing "Message delivery status: Acknowledged", the overlay could display:

```
Message delivery status: Acknowledged
First received by: [Node Name/ID]
Distance: [X km/miles] (if location available)
Signal: [SNR/RSSI values] (if available)
Time: [Acknowledgment timestamp]
[View on Map] (if node location is known)
```

This enhancement would be particularly valuable for users in sparse network areas who are actively working to expand mesh coverage and need detailed feedback about their transmission success.

[image]

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

moverhorst-pitcrew, 2025-08-12:
> Don't trust AI.
> 
> My opinion: it can't be done without MQTT.
> 
> Currently, we only have enough information to tell:
> 
> * a message was rebroadcasted by a node (implicit ack, we don't know who)
> * an ack was received from a target node (direct messaging stuff, has source id)

There's always a way haha, but it might not be pretty hehe...
Can't imagine MQTT being part of Meshtastics long-term future, so anything that relies on it, feels a bit like a temporary workaround.

Fabryz, 2025-10-03:
During all my tests this feature would be very helpful, at the moment my workaround is to block nodes from other certain nodes, so IF/WHEN I get a confirmation I know for sure that they weren't my nodes.

Still impractical to me if you ask :/