meshtastic/firmware #7599: [Bug]: The firmware does not start on the esp32 super mini and E220-900MM22S

rapid32, 2025-08-10:
### Category

Other, Hardware Compatibility

### Hardware

DIY

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.5.7505fe7a7

### Description

I compiled the.../diy/esp32c3_super_mini assembly platform and poured it into a homemade device mounted on an esp32 super mini, E220-900MM22S.

Прошивка не стартует нужным образом, прошу помощи!

Contents of the variant.h:

#ifndef _VARIANT_ESP32C3_SUPER_MINI_
#define _VARIANT_ESP32C3_SUPER_MINI_

/*----------------------------------------------------------------------------
 *        Headers
 *----------------------------------------------------------------------------*/

#ifdef __cplusplus
extern "C" {
#endif // __cplusplus

// I2C (Wire) & OLED
//#define WIRE_INTERFACES_COUNT (1)
//#define I2C_SDA (1) 
//#define I2C_SCL (0)   

#define USE_SSD1306   

// GPS
#undef GPS_RX_PIN
#undef GPS_TX_PIN
#define GPS_RX_PIN (0)
#define GPS_TX_PIN (9)   

// Button
#define BUTTON_PIN (8) // BOOT button

// LoRa
#define USE_LLCC68
#define USE_SX1262 
// #define USE_RF95
#define USE_SX1268

#define LORA_DIO0 RADIOLIB_NC
#define LORA_RESET (1)
#define LORA_DIO1 (3)
#define LORA_RXEN (2)  
#define LORA_BUSY (10)  
#define LORA_SCK (4)    
#define LORA_MISO (5)  // rf sniffer
#define LORA_MOSI (6)   
#define LORA_CS (7)

#define SX126X_CS LORA_CS
#define SX126X_DIO1 LORA_DIO1
#define SX126X_BUSY LORA_BUSY
#define SX126X_RESET LORA_RESET
#define SX126X_RXEN LORA_RXEN

#define SX126X_DIO3_TCXO_VOLTAGE (1.8)
#define TCXO_OPTIONAL // make it so that the firmware can try both TCXO and XTAL

#ifdef __cplusplus
}
#endif

/*----------------------------------------------------------------------------
 *        Arduino objects - C++ only
 *----------------------------------------------------------------------------*/

#endif

### Relevant log output

```Shell
ESP-R�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c

=> 0x403cc710: ?? ??:0
load:0x403ce710,len:0x2624

=> 0x403ce710: ?? ??:0
entry 0x403cc710

=> 0x403cc710: ?? ??:0
E (356) esp_core_dump_flash: No core dump partition found!
E (357) esp_core_dump_flash: No core dump partition found!
[   506][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=1 scl=0 freq=100000
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x42044a6f

=> 0x42044a6f: BleOta::findEspOtaAppPartition() at ??:?
file: "src/platform/esp32/BleOta.cpp" line 16
func: static const esp_partition_t* BleOta::findEspOtaAppPartition()
expression: esp_ota_get_partition_description(part, &app_desc)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42045e37
ot exist, no permits for creation
[   604][E][vfs_api.cpp:105] open(): /littlefs/prefs/db.proto does not exist, no permits for creation
[   654][E][vfs_api.cpp:105] open(): /littlefs/prefs/config.proto does not exist, no permits for creation
[   673][E][vfs_api.cpp:105] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
[   708][E][vfs_api.cpp:105] open(): /littlefs/prefs/uiconfig.proto does not exist, no permits for creation
[   838][E][vfs_api.cpp:105] open(): /littlefs/prefs/cannedConf.proto does not exist, no permits for creation
```

rapid32, 2025-08-11:
> Did you attach a USB-TTL adapter to GPIO20/21to see the actual debug logs from meshtastic?

Yes. Also, I use a USB-TTL adapter to upload firmware.

mverch67, 2025-08-11:
> Yes. Also, I use a USB-TTL adapter to upload firmware.

Can you run a serial terminal on that USB-TTL device to see the debug log?

rapid32, 2025-08-12:
> > Да. Кроме того, я использую адаптер USB-TTL для загрузки прошивки.
> 
> Можно ли запустить последовательный терминал на этом устройстве USB-TTL, чтобы просмотреть журнал отладки? Можете ли вы поделиться этим журналом?

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c

=> 0x403cc710: ?? ??:0
load:0x403ce710,len:0x2624

=> 0x403ce710: ?? ??:0
entry 0x403cc710

=> 0x403cc710: ?? ??:0
E (356) esp_core_dump_flash: No core dump partition found!
E (357) esp_core_dump_flash: No core dump partition found!
[   506][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=1 scl=0 freq=100000
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x42044a6f

=> 0x42044a6f: BleOta::findEspOtaAppPartition() at ??:?
file: "src/platform/esp32/BleOta.cpp" line 16
func: static const esp_partition_t* BleOta::findEspOtaAppPartition()
expression: esp_ota_get_partition_description(part, &app_desc)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42045e37
ot exist, no permits for creation
[   604][E][vfs_api.cpp:105] open(): /littlefs/prefs/db.proto does not exist, no permits for creation
[   654][E][vfs_api.cpp:105] open(): /littlefs/prefs/config.proto does not exist, no permits for creation
[   673][E][vfs_api.cpp:105] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
[   708][E][vfs_api.cpp:105] open(): /littlefs/prefs/uiconfig.proto does not exist, no permits for creation
[   838][E][vfs_api.cpp:105] open(): /littlefs/prefs/cannedConf.proto does not exist, no permits for creation

rapid32, 2025-08-13:
> This is the esp-idf log but we want to see the meshtastic log. Try setting `ARDUINO_USB_MODE=0` or `ARDUINO_USB_CDC_ON_BOOT=0` in the .ini and check if the proper log appears.

Everything worked, thanks for your help!

Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
E (355) esp_core_dump_flash: No core dump partition found!
E (355ʔ▒@INFO  | ??:??:?? 0

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (16384/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (86 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (172 Bytes)
DEBUG | ??:??:?? 0  /prefs/device.proto (172 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (100 Bytes)
DEBUG | ??:??:?? 0  /prefs/nodes.proto (314 Bytes)
[   423][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=19 scl=1                                                                                                                                                             8 freq=100000
DEBUG | ??:??:?? 0 Power::lipoInit lipo sensor is not ready yet
INFO  | ??:??:?? 0 Scan for i2c devices
DEBUG | ??:??:?? 0 Scan for I2C devices on port 1
INFO  | ??:??:?? 0 No I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 S:B:255,2.7.5.7505fe7a7,esp32c3_super_mini,meshtastic/firmwar                                                                                                                                                             e
DEBUG | ??:??:?? 0 Total heap: 232116
DEBUG | ??:??:?? 0 Free heap: 199968
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
DEBUG | ??:??:?? 0 NVS: UsedEntries 98, FreeEntries 532, AllEntries 630, NameSpa                                                                                                                                                             ces 5
DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 0 Number of Device Reboots: 33
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4                                                                                                                                                             204494d
file: "src/platform/esp32/BleOta.cpp" line 16
func: static const esp_partition_t* BleOta::findEspOtaAppPartition()
expression: esp_ota_get_partition_description(part, &app_desc)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0                                                                                                                                                             x42045d15
file: "src/platform/esp32/BleOta.cpp" line 30
func: static String BleOta::getOtaAppVersion()
expression: esp_ota_get_partition_description(part, &app_desc)
INFO  | ??:??:?? 0 No BLE OTA firmware available
INFO  | ??:??:?? 0 No WiFi OTA firmware available
INFO  | ??:??:?? 0 Init NodeDB
[   581][E][vfs_api.cpp:105] open(): /littlefs/static/static does not exist, no                                                                                                                                                              permits for creation
[   595][E][vfs_api.cpp:105] open(): /littlefs/prefs/db.proto does not exist, no                                                                                                                                                              permits for creation
INFO  | ??:??:?? 0 Load /prefs/nodes.proto
INFO  | ??:??:?? 0 Loaded /prefs/nodes.proto successfully
INFO  | ??:??:?? 0 Loaded saved nodedatabase version 24, with nodes count: 1
INFO  | ??:??:?? 0 Load /prefs/device.proto
INFO  | ??:??:?? 0 Loaded /prefs/device.proto successfully
INFO  | ??:??:?? 0 Loaded saved devicestate version 24
INFO  | ??:??:?? 0 Load /prefs/config.proto
INFO  | ??:??:?? 0 Loaded /prefs/config.proto successfully
INFO  | ??:??:?? 0 Loaded saved config version 24
INFO  | ??:??:?? 0 Load /prefs/module.proto
INFO  | ??:??:?? 0 Loaded /prefs/module.proto successfully
INFO  | ??:??:?? 0 Loaded saved moduleConfig version 24
INFO  | ??:??:?? 0 Load /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 0 Loaded saved channelFile version 24
[   718][E][vfs_api.cpp:105] open(): /littlefs/prefs/uiconfig.proto does not exi                                                                                                                                                             st, no permits for creation
ERROR | ??:??:?? 0 Could not open / read /prefs/uiconfig.proto
DEBUG | ??:??:?? 0 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 0 Use nodenum 0xb0d0993c
[... 59 lines cut ...]
INFO  | ??:??:?? 1 numChannels: 2 x 250.000kHz
INFO  | ??:??:?? 1 channel_num: 1
INFO  | ??:??:?? 1 frequency: 868.825012
INFO  | ??:??:?? 1 Slot time: 12 msec
INFO  | ??:??:?? 1 Final Tx power: 20 dBm
INFO  | ??:??:?? 1 SX126x init result 0
INFO  | ??:??:?? 1 Frequency set to 868.825012
INFO  | ??:??:?? 1 Bandwidth set to 250.000000
INFO  | ??:??:?? 1 Power output set to 20
DEBUG | ??:??:?? 1 Current limit set to 140.000000
DEBUG | ??:??:?? 1 Current limit set result 0
DEBUG | ??:??:?? 1 Set DIO2 as not RF switch, result: 0
DEBUG | ??:??:?? 1 SX126X_TXEN not defined, defaulting to RADIOLIB_NC
[... cut ...]
INFO  | ??:??:?? 1 Set RX gain to boosted mode; result: 0
INFO  | ??:??:?? 1 LLCC68 init success
INFO  | ??:??:?? 1 Not using WIFI
DEBUG | ??:??:?? 1 LoRA bitrate = 394.915253 bytes / sec
INFO  | ??:??:?? 1 PowerFSM init, USB power=1
DEBUG | ??:??:?? 1 State: BOOT
DEBUG | ??:??:?? 1 Free heap  :  161684 bytes
DEBUG | ??:??:?? 1 Free PSRAM :       0 bytes
DEBUG | ??:??:?? 1 Node status update: 1 online, 1 total
INFO  | ??:??:?? 1 [RangeTest] Range Test Module - Disabled
INFO  | ??:??:?? 4 [PowerFSM] Init the NimBLE bluetooth module