meshtastic/firmware #7600: [Bug]: WantAck/PKI Retries Under Asymmetric Ingress/Egress Conditions - UDP PKI Failure?

daytonmesh, 2025-08-11:
### Category

Other

### Hardware

T-Beam

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.33

### Description

### Suspected Root Cause

The firmware likely isn't properly skipping decoding for inbound PKI packets on UDP, treating them like PSK and failing when hash=0x0.  Under asymmetric Ingress/Egress conditions (UDP ingress, RF egress; or likely even symmetric UDP-only conditions, though I haven't tested this), this causes the receiving node to refuse DMs twice before falling back to flood routing, causing a roughly ~17 second delay in DM delivery, and harming the intended function of retry logic for less-than-ideal link conditions.

### Detailed Description

**Application Outline**

I'm working on a backhaul application called gemesh (Golang Ephemeral Mesh) that leverages the new(ish) UDP Multicast functionality in order to extend my local mesh to _topographically challenged_ regions of my metro area.  

This backhaul application runs on Pi Zero 2 W's or similar SBCs/Linux instances local to and on the same subnet as a gateway node, establishes a UDP connection to said node, and establishes a TLS connection to a private MQTT server; brokering data between gateway nodes in a hub/spoke pub/sub architecture.  The idea being to extend UDP transparently across WAN, thereby extending the mesh.

**Test Setup**

Test setup consists of four nodes; two gateways (DTC1 and DTC2), each with their own instance of gemesh, two client nodes (SGC1 and SGM1), and a private MQTT server.  Each gateway node and its gemesh instance is running on its own VLAN in order to segregate UDP traffic and simulate real-world deployment.

All four nodes have the same, private, primary channel, with the same PSK.  Both Gateways are running 2.6.33 firmware.

**Aberrant Test Scenario**

The gemesh/gateway system is working as expected in all but one test scenario designed to simulate an asymmetric RF link.  In this specific test scenario, 3 of the four nodes are used; both gateways (DTC1 and DTC2), one client (SGC1).  All nodes are configured for the same frequency slot and primary channel, and all nodes are well within reliable RF range of each other (they're all at my house).

In this scenario DTC1's TX is disabled to simulate a condition in which a Gateway can hear other nodes via RF, but cannot be heard over RF.  The test is initiated by sending a direct message (implicating WantAck/PKI/ReliableRouter) from DTC1 to SGC1.  In this test, the expected (and only possible) path is DTC1 -> UDP -> gemesh -> MQTT -> gemesh -> UDP -> DTC2 -> LoRa RF -> SGC1.  Running traceroute reliably confirms this path (though slowly, as I'll get into).

**Test Outcomes**

Upon running this test, the same behavior happens reliably.  I initiate the Direct Message from DTC1 to SGC1, while watching the gemesh logs of both DTC1 and DTC2 in real-time.  The logs (shared below) indicate that gemesh is working as expected; DTC1 sends the packet over UDP, the gemesh UDP listener loop grabs this packet and sends it via MQTT to all other gateway nodes.  The gemesh DTC2 instance hears this MQTT message and forwards it, via UDP, to the DTC2 node.

However, no response/rebroadcast is transmitted via UDP from DTC2 back to DTC1 (DTC2's serial output shows it trying to decrypt the packet and failing).  As such, DTC1's retry logic begins to execute.  ~7 seconds pass, a second attempt is sent from DTC1, and again, DTC2's gemesh instance receives it, it is forwarded to UDP, and the DTC2 node fails to respond (More decryption/channel errors on serial output).  Another ~7 seconds pass, DTC1 attempts for a third time, and this time DTC2 falls back to flood routing, and a reply is sent over UDP.  DTC1 presumably hears the rebroadcast, but it's out of retries anyway, so it's somewhat irrelevant.  Very shortly after, an ACK packet is sent from SGC1, which makes its way through the backhaul and DTC1 gets its confirmation of delivery.

**Problem Summary**

So to summarize, DTC1 is the DM sender with TX disabled, SGC1 the recipient.  Three retries, success on the third retry.  This is likely an inbound UDP PKI packet problem.

When sending from SGC1 to DTC1, delivery and ACK reliably happens on the first attempt.  Also, it only happens for PKI/WantAck packets.  When I send a message on a shared channel from DTC1, delivery happens immediately.  Along with the logs, this points strongly to a UDP PKI problem.

Other test scenarios are working great.  When everything is on the same frequency slot/preset, all works as expected and messages are delivered immediately.  It's also fine bridging different frequency slots, etc.  It's only in this asymmetric link regime, and only inbound DM via UDP, where things start borking.

**DTC1 gemesh logs illustrating retries and eventual success:**

```
[Initial Attempt]
2025/08/10 01:04:05.283795 [UDP → MQTT] 10.10.20.200:4403:dmgwtc1 → ALL - Packet ID: 1ef60ad146140fdce6592847d7169a78a16431b8c744766e666eed774ee95b65
[Initial Attempt MQTT Fan-in Protection]
2025/08/10 01:04:05.299678 Dropping Self-Generated MQTT Packet UUID: 5d515959-4166-475c-af39-b6bafbbe5d1e
[Second Attempt]
2025/08/10 01:04:12.181784 [UDP → MQTT] 10.10.20.200:4403:dmgwtc1 → ALL - Packet ID: b6a3c8966a6dba9ebe2808ac7ae3e42b93135c701a5f059d4258d0b7db2b65a6
[Second Attempt MQTT Fan-in Protection]
2025/08/10 01:04:12.197283 Dropping Self-Generated MQTT Packet UUID: edfaeb6a-ccee-4bb9-937f-dcadfed1cd05
[Third Attempt]
2025/08/10 01:04:19.223307 [UDP → MQTT] 10.10.20.200:4403:dmgwtc1 → ALL - Packet ID: 2afcb259eb85f69a3da9d199715a100d9111ddc32dcccc62cdc833a9b9fc7840
[Third Attempt MQTT Fan-in Protection]
2025/08/10 01:04:19.238670 Dropping Self-Generated MQTT Packet UUID: e352b559-b6a2-45f5-8e8c-1aef6f9fbb46
[Likely Flood Rebroadcast from DTC2]
2025/08/10 01:04:19.520177 [MQTT → UDP] dmgwtc2 → dmgwtc1 - Packet ID: f9a45c04bde03cd6784a724565d95620a8dd4e97397ddddca34f01312f8b6fad
[Likely ACK from SGC1 - Message appears on client app at this point]
2025/08/10 01:04:21.236540 [MQTT → UDP] dmgwtc2 → dmgwtc1 - Packet ID: 99ade72e98630292a2687f4db2e1cd00a8309d4ba0c3da7a9ec4a674aa957c17
```

**DTC2 gemesh logs illustrating retries and eventual success:**

```
[Initial Attempt Sent to DTC2 via UDP]
Aug 10 05:04:05 dmgatewaytc2 gemesh[20828]: 2025/08/10 05:04:05.344896 [MQTT → UDP] dmgwtc1 → dmgwtc2 - Packet ID: 1ef60ad146140fdce6592847d7169a78a16431b8c744766e666eed774ee95b65
[Second Attempt Sent to DTC2 via UDP]
Aug 10 05:04:12 dmgatewaytc2 gemesh[20828]: 2025/08/10 05:04:12.207776 [MQTT → UDP] dmgwtc1 → dmgwtc2 - Packet ID: b6a3c8966a6dba9ebe2808ac7ae3e42b93135c701a5f059d4258d0b7db2b65a6
[Third Attempt Sent to DTC2 via UDP]
Aug 10 05:04:19 dmgatewaytc2 gemesh[20828]: 2025/08/10 05:04:19.341679 [MQTT → UDP] dmgwtc1 → dmgwtc2 - Packet ID: 2afcb259eb85f69a3da9d199715a100d9111ddc32dcccc62cdc833a9b9fc7840
[Likely DTC2 Flood Rebroadcast]
Aug 10 05:04:19 dmgatewaytc2 gemesh[20828]: 2025/08/10 05:04:19.498256 [UDP → MQTT] 10.10.70.20:4403:dmgwtc2 → ALL - Packet ID: f9a45c04bde03cd6784a724565d95620a8dd4e97397ddddca34f01312f8b6fad
[Rebroadcast MQTT Fan-in Protection]
Aug 10 05:04:19 dmgatewaytc2 gemesh[20828]: 2025/08/10 05:04:19.513264 Dropping Self-Generated MQTT Packet UUID: bc1670c8-488e-47c2-8890-2a18e8e5d415
[Likely SGC1 ACK]
Aug 10 05:04:21 dmgatewaytc2 gemesh[20828]: 2025/08/10 05:04:21.199831 [UDP → MQTT] 10.10.70.20:4403:dmgwtc2 → ALL - Packet ID: 99ade72e98630292a2687f4db2e1cd00a8309d4ba0c3da7a9ec4a674aa957c17
```

**Serial Output Logs:**
_Note: shared key prefixed have been sterilized._

[dtc1_log.txt](https://github.com/user-attachments/files/21708243/dtc1_log.txt)

[dtc2_log.txt](https://github.com/user-attachments/files/21708244/dtc2_log.txt)

**Packet Flow Illustration**

[image]

### Relevant log output

```Shell

```

daytonmesh, 2025-08-12:
Perhaps this is relevant: https://github.com/meshtastic/firmware/blob/master/src/mesh/udp/UdpMulticastHandler.h#L56-L58

lkosson, 2025-08-12:
Your `dtc2_log.txt` shows that initial packet comes in with `nextHop=0xb8` while - I assume - your `DTC2` nodeId ends with `b8`, so it ignores 1st and 2nd transmission. On third broadcast `DTC1` reverts to flooding (there is no `nextHop` reported in log) so `DTC2` accepts and rebroadcasts the packet.

Perhaps from your earlier tests `DTC1` learned that `ba3f69b8` is directly reachable so it sets `nextHop` to `0xb8` causing `DTC2` to ignore it.

lkosson, 2025-08-13:
Those two fields of `MeshPacket` - `pki_encrypted` and `public_key` - are not transmitted over-the-wire or over-the-radio. They are only kept internally on one device for bookkeeping.

I suggest carefully inspecting `relay_node` field of the ACK packet sent from `SGC1` as seen inbound and outbound at `DTC2` node. This packet, when matched with corresponding request packet according to `NextHopRouter::sniffReceived`, sets `next_hop` for subsequent transmissions to destination node.

`next_hop` is not updated, unless ACK came the same way as original packet (see `sniffReceived` above) unless ACK comes directly from destination (i.e. `hop_start` == `hop_limit`). Perhaps packets you are tunneling somehow don't have their `relay_node` updated, `DTC1` sees them as sent by `SGC1` and therefore thinks it is directly reachable?

Or maybe in your case `SGC1` can reach `DTC1` directly, but `DTC1` needs to relay via `DTC2` to reach `SGC1`? In that case `sniffReceived` might store `next_hop` for the node. But still it shouldn't matter that packets were sent as UDP on the way or if they are PKI-encrypted, only thing important is that they were qualified for `NextHopRouter`.

daytonmesh, 2025-08-14:
> Or maybe in your case SGC1 can reach DTC1 directly, but DTC1 needs to relay via DTC2 to reach SGC1?

That is correct.  In this test case, DTC1's TX is intentionally disabled in order to simulate an asymmetric link scenario.  SGC1 can transmit directly to DTC1, but DTC1's only outbound path to SGC1 is through the backhaul to DTC2.

In this trace route screen shot, it's SGM1 instead of SGC1, but it's the same for both SGC1 and SGM1.

[image]

lkosson, 2025-08-14:
Then it is this condition that causes `DTC1` to learn to `next_hop` directly to `SGM1`:

https://github.com/meshtastic/firmware/blob/52f0e5a3db06a7a3c77dade181c26c508faa2e11/src/mesh/NextHopRouter.cpp#L77

Whether it is good or bad is for project maintainers to decide. In both cases some edge cases will be misinterpreted.