meshtastic/firmware #7623: [Bug]: mqtt cant enable tls in 2.7.5

shortwavesurfer2009, 2025-08-14:
### Category

Other

### Hardware

Seeed Card Tracker T1000-E

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.5

### Description

I have firmware 2.7.5 installed on my device and if I try to enable TLS for MQTT on the default server I get

"Invalid MQTT config: default server address must not have a port specified, Window Meshtastic"

### Relevant log output

```Shell

```

thebentern, 2025-08-19:
@shortwavesurfer2009 did you experience this on Android, iOS or python CLI?

shortwavesurfer2009, 2025-08-19:
> [@shortwavesurfer2009](https://github.com/shortwavesurfer2009) did you experience this on Android, iOS or python CLI?

Android

thebentern, 2025-08-19:
Thanks! I'll take a look