meshtastic/firmware #7630: [Bug]: T-Echo (w/ BME280) - Display backlight not working

EnygmaINC, 2025-08-15:
### Category

Hardware Compatibility

### Hardware

T-Echo

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.5.ddd1499

### Description

After flashing to 2.7.5.ddd1499 (Alpha, capacitive touch button at top of body no longer works, and backlight for E-Ink display no longer turns on.

### Relevant log output

```Shell

```

ViezeVingertjes, 2025-08-15:
Highly likely its as simple as this; but can't test it right now. if anyone could; feel free to PR it after testing (add a comment here so we dont do double work), otherwise i will test it a.s.a.p. when i get the chance.

```
Index: src/graphics/niche/Inputs/TwoButton.cpp
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/src/graphics/niche/Inputs/TwoButton.cpp b/src/graphics/niche/Inputs/TwoButton.cpp
--- a/src/graphics/niche/Inputs/TwoButton.cpp	(revision c8694f9f2d6a6bd7b38afd421b753f5faaeb644d)
+++ b/src/graphics/niche/Inputs/TwoButton.cpp	(date 1755261429390)
@@ -97,7 +97,7 @@
 
     assert(whichButton < 2);
     buttons[whichButton].pin = pin;
-    buttons[whichButton].activeLogic = LOW; // Unimplemented
+    buttons[whichButton].activeLogic = internalPullup ? HIGH : LOW;
 
     pinMode(buttons[whichButton].pin, internalPullup ? INPUT_PULLUP : INPUT);
 }
Index: variants/nrf52840/t-echo/nicheGraphics.h
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/variants/nrf52840/t-echo/nicheGraphics.h b/variants/nrf52840/t-echo/nicheGraphics.h
--- a/variants/nrf52840/t-echo/nicheGraphics.h	(revision c8694f9f2d6a6bd7b38afd421b753f5faaeb644d)
+++ b/variants/nrf52840/t-echo/nicheGraphics.h	(date 1755262758211)
@@ -99,7 +99,7 @@
     // #1: Aux Button (Capacitive Touch Button)
     // - short: momentary backlight
     // - long: latch backlight on
-    buttons->setWiring(1, PIN_BUTTON_TOUCH);
+    buttons->setWiring(1, PIN_BUTTON_TOUCH, true);
     buttons->setTiming(1, 50, 5000); // 5 seconds before latch - limited by T-Echo's capacitive touch IC
 
     buttons->setHandlerDown(1, [inkhud, backlight]() {
```

EnygmaINC, 2025-08-17:
**NOTE**: It seems to only happen with **BaseUI (no InkHUD)** version of the Alpha.

ViezeVingertjes, 2025-08-19:
> **NOTE**: It seems to only happen with **BaseUI (no InkHUD)** version of the Alpha.

Yes, its in shared code. Havent had the time yet, but its still on my TODO list.

EnygmaINC, 2025-08-19:
Awesome, appreciate the update!

ViezeVingertjes, 2025-08-20:
https://drive.google.com/file/d/1H2CnqgqROVJPleQ2gV3Nuv8WNqzBy_o8/view?usp=drive_link

That's the best i could do right now.. 
There is some delay at certain times, which might be the waiting on transmission to end, as that also triggered the touch button..., but will look into that later before i create a PR.

EnygmaINC, 2025-08-29:
> https://drive.google.com/file/d/1H2CnqgqROVJPleQ2gV3Nuv8WNqzBy_o8/view?usp=drive_link
> 
> That's the best i could do right now.. There is some delay at certain times, which might be the waiting on transmission to end, as that also triggered the touch button..., but will look into that later before i create a PR.
> 
> The draft can be found at [#7699](https://github.com/meshtastic/firmware/pull/7699)
> 
> I do need to ask; why aren't you using InkHUD? It's so much faster/responsive. [@EnygmaINC](https://github.com/EnygmaINC)

Actually, for navigating through menus - InkHUD is faster and more responsive; button pushes and all, BaseUI is slower by a millisecond (think, pushing button to go down a list of menu options). 

And, the backlight thing is a *HUGE* showstopper for me. BaseUI's options for turning on the backlight are difficult to find on a T-ECHO's screen in the dark, and there isn't an easy "SCREEN" settings sub-menu.

RCG-RCorrales, 2025-11-04:
Still open

viperet, 2026-01-06:
I created a PR that gives the touch button in BaseUI the same function as in InkHUD - short touch enables backlight for 3 sec, long touch toggles backlight on/off. 
https://github.com/meshtastic/firmware/pull/9198

viperet, 2026-01-13:
@EnygmaINC If you know how to build firmware, you can clone my fork, build a custom firmware for your T-Echo, and test backlight control. Would be grateful for your feedback on the PR.

EnygmaINC, 2026-01-13:
> [@EnygmaINC](https://github.com/EnygmaINC) If you know how to build firmware, you can clone my fork, build a custom firmware for your T-Echo, and test backlight control. Would be grateful for your feedback on the PR.

I would if I could, but unfortunately building firmware is not in my skill set.

closed 2026-03-22 by thebentern