meshtastic/firmware #7680: [Feature Request]: Add Israel region (AS923-4 / 917–920 MHz) to lora.region enum (missing in 2.6.11)

fr0stnk, 2025-08-19:
### Platform

other

### Description

Hi,
The Israel LoRa/meshtastic band is **AS923-4 (917–920 MHz)**. It’s officially recognized by the LoRa Alliance and by Israel’s Ministry of Communications (MoC).
In current firmware (tested on 2.6.11.60ec05e, T-Echo InkHUD), `lora.region` does **not** include `AS923_4`, so users in Israel must hack around with `lora.override_frequency`. Please add a first-class region.

**Current behavior**
``` > meshtastic --port COM5 --set lora.region AS923_4
lora.region does not have an enum called AS923_4
Choices: ANZ, ANZ_433, BR_902, CN, EU_433, EU_868, IN, JP, KR, KZ_433, KZ_863,
LORA_24, MY_433, MY_919, NP_865, NZ_865, PH_433, PH_868, PH_915, RU, SG_923,
TH, TW, UA_433, UA_868, UNSET, US
```

**Expected behavior**
- lora.region includes AS923_4 (or IL_917_920), selectable from CLI and UI
- Region constrains frequencies to 917.0–920.0 MHz and provides sensible default channels inside that window

**Region (Israel)**

| Region Code | Description | Frequency Range (MHz) | Duty Cycle (%) | Power Limit (dBm) |
| ------------- | ------------- | ------------- | ------------- | ------------- |
| AS923_4 | Israel (LoRaWAN/Meshtastic) | 917.0 - 920.0 | 1 (data) / 10 (control) | 14 (data ERP) / 27 (control ERP) |

Government acts / primary sources (URLs):
1) Israel Ministry of Communications – 2017 public notice (pilot): 915–917 MHz IoT/LoRa
https://www.gov.il/BlobFolder/news/2112017/he/הודעה%20לציבור-שימוש%20בפס%20התדרים%20915%20עד%20917%20מהץ.pdf
2) Wireless Telegraph (Type Approvals) Regulations, 2021 – RESHUMOT (official gazette) PDF
   (contains the 917–920 MHz LoRaWAN section with duty-cycle & ERP limits)
   https://www.chamber.org.il/media/164960/תקנות-הטלגרף-האלחוטי-אישורי-התאמה-פורסמו-ברשומות-ב-2932021-קובץ-תקנות-9301.pdf
3) Consolidated text (reference): Nevo legal database – “תקנות הטלגרף האלחוטי (אישורי התאמה), תשפ״א–2021”
   (shows the same 917–920 MHz constraints incl. 1%/10% duty cycle and out-of-band emission limits)
   https://www.nevo.co.il/law_html/law01/502_483.htm

closed 2025-08-19 by garthvh

fr0stnk, 2025-08-19:
@garthvh 
Thanks for the response!

I'm a bit confused though, Meshtastic already supports `UA_868 (Ukraine)`, which explicitly has a 1% duty cycle according to the official [LoRa Region Settings docs](https://meshtastic.org/docs/configuration/radio/lora/)

| Region Code | Duty Cycle (%) | Power Limit (dBm) |
| -------------- | ---------------- | -------------------- |
| UA_868: 868.0 – 868.6 MHz | 1% duty cycle | 14 dBm | 

So my question is:
- Why was UA_868 added if 1% duty cycle regions are not supported?
- Is there a technical limitation that prevents adding Israel with 1% duty cycle, or was the choice more about policy / maintainability?

Israel's allocation at 915–917 MHz with 1% duty cycle, 27 dBm is comparable in constraints to UA_868. So, if UA_868 works, it seems Israel could be added in the same manner.

Can you clarify why 1% duty cycle is acceptable for UA_868 but not for Israel?

fifieldt, 2025-08-29:
1% is too little, even 10% regions have issues. UA_868 is historical.