meshtastic/firmware #7687: [Feature Request]: Enhance API - GPS handling

ftrueck, 2025-08-19:
### Platform

Cross-Platform

### Description

It would be fantastic if the API would be enhanced to support switching off/on the GPS functionality without rebooting the device. For example the the feature to triple-click the user button to disable/enable GPS is superb. If this functionality would be available via the API it would allow several convenient features like for example the real-time tracking in the map in MUI. Also the MUI would benefit in disable/enable the GPS this way. Currently it is not possible to enable/disable GPS via a touch of the GPS button. Long pressing the button only sends the API command to change GPS config which results in a reboot. On a Lilygo T-Deck /Plus this takes a very long time. I know base UI supports disable/enable of GPS in realtime but the lack of the map features makes this a less preferable option for the T-Deck devices and also other devices that have a color screen and are capable of displaying a map e.g.

thebentern, 2025-08-19:
Great idea! I think this actually may be possible with the new inject input broker event admin command, but I need to confirm that.

mverch67, 2025-08-19:
Regarding GPS handling I would like to propose to send the GPS position locally over the API based on _position.gps_update_interval_ (e.g. 120s) and not as currently based on _position.position_broadcast_secs_ (900) similar to the local battery updates that are sent every 60s to phone. This way MUI could update the GPS position more often than just every 15 minutes or whatever value is configured for broadcast. The local updates could also include the exact position and not the precision castrated value.

thebentern, 2025-08-19:
> Regarding GPS handling I would like to propose to send the GPS position locally over the API based on _position.gps_update_interval_ (e.g. 120s) and not as currently based on [...]

Good idea. I think we can do this in a new protobuf message so that there is no ambiguity or confusion between the two sources

closed 2025-10-17 by github-actions[bot]