meshtastic/firmware #7690: [Feature Request]: Node-side Configuration for BLE MTU size & Data Length Entension

michaelmedellin, 2025-08-20:
### Platform

Cross-Platform

### Description

iOS family apps communicating with BLE peripherals do not have the ability to request MTU sizes or enable BLE Data Length Extension, leaving it to peripherals to negotiate with the Core Bluetooth central manager instance. 

The Device API developer docs explicitly asks for clients to request a 512 MTU, but the iOS clients can't do this today. I'm experimenting with a custom firmware build to enable this, but trying to gauge community/maintainer interest in a contribution to help address this.

Below is sample code I'm using successfully on a Station G2 and Heltec v3, using the NimbleBluetooth framework. 

---- 

Example, functioning code for me working on `2.6.11` with 512 MTU on iOS (vs. 252 today). Can confirm with `peripheral.maximumWriteValueLength(for: .withoutResponse)` in Core Bluetooth.

```cpp
void NimbleBluetooth::setup()
{
    NimBLEDevice::setMTU(512);
```

```cpp
virtual void onConnect(NimBLEServer *pServer)
    {
        LOG_INFO("BLE device connected");
       
        std::vector<uint16_t> connHandles = pServer->getPeerDevices();
        if (!connHandles.empty()) {
            uint16_t connHandle = connHandles[0];
            
            int result = ble_gap_set_data_len(connHandle, 251, 2120); 
            if (result == 0) {
                LOG_INFO("DLE enabled: 251 bytes per packet");
            } else {
                LOG_WARN("DLE configuration failed: %d", result);
            }
        }
        
        bluetoothStatus->updateStatus(new meshtastic::BluetoothStatus(meshtastic::BluetoothStatus::ConnectionState::CONNECTED));
    }
```

Maybe this could be configured/enabled/disabled through the `BluetoothConfig` protobuf? 

Cheers.

garthvh, 2025-08-20:
Does not seem like it should be a user configurable option, what is the advantage of these settings?

garthvh, 2025-08-20:
Yeah that documentation is ancient, probably from before the iOS app even existed, iOS can't do 512 mtu at all.

closed 2025-08-20 by garthvh