meshtastic/firmware #7714: [Bug]: T-Deck Pro Touchscreen stops working after sending message with keyboard

wifiguru10, 2025-08-22:
### Category

Other

### Hardware

T-Deck

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [x] Meshtastic UI aka MUI colorTFT
- [x] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.6

### Description

Built 2.7.6 firmware from source and flashed T-Deck Pro (4G version)

Description: 
Touchscreen stops working after sending a message with the keyboard. UI is completely unresponsive but does update occasionally. You can still use the device via BLE. Device needs to be reset

If you respond to the message with a preset message, it keeps working. The UI only breaks if you send a message with the keyboard.

Replication:
1. Reset device
2. receive a message
3. reply to message with "freetext" option
4. after hitting "enter" the touchscreen stops working

Workaround:
1. The only way to recover from touchscreen not working is to reboot (reset switch)

### Relevant log output

```Shell

```

Xaositek, 2025-08-29:
@HarukiToreda or @jp-bennett - may need your assistance on this one. I was able to reproduce this fairly reliably.

thebentern, 2025-08-29:
I repro'd this on T-Deck Pro, T-Deck and T-LoRa Pager, so there is a logic bug somewhere. Seems to be only after freetext and does not seem to be affected by Canned Messages.

Xaositek, 2025-08-30:
Fixed in version 2.7.7

closed 2025-08-30 by Xaositek