meshtastic/firmware #7734: [Bug]: No air pressure reading for BME680 sensor in Android app

OTLabs, 2025-08-24:
### Category

Hardware Compatibility

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.5.ddd1499

### Description

I have BME680 sensor connected to LilyGO T3-S3 v1.2 node with 2.7.5.ddd1499 Alpha firmware flashed.

The screen of the node successfully shows complete telemetry data (temperature, humidity, IAQ, and air pressure) while the connected Android application (v2.6.30) only shows the readings of temperature, humidity and IAQ, missing to show the readings for air pressure.

The nodes connected over mesh are also unable to show air pressure readings from this node.

Initially this issue was reported as https://github.com/meshtastic/Meshtastic-Android/issues/2826 , but later I was suggested that actually that could be related with firmware.

### Relevant log output

```Shell

```

RicInNewMexico, 2025-08-24:
And you've confirmed that you have the environment metrics module enabled, not just the onscreen display of environment metrics? 

Without a log snippet attached to the issue I have to ask because this would cause the same symptoms and it's happened before.

DaneEvans, 2025-08-31:
Close. Once the issue was fully communicated, it's an app bug.